micropython/micropython-esp32 #136: make fails "xtensa-esp32-elf-gcc: error: build/bootloader//home/robert/esp/esp-idf//components/bootloader/src/main/bootloader_start.o: No such file or directory"

TrifidSagitarius, 2017-07-22:
Hello, I'm trying to install micropython in my esp32 (DoIT) but I keep getting the following error.
My setup here is ubuntu 17.04 and I have xtensa and espressif installed, I have no problems flashing the esp32 using espressif.

Path=home/robert/esp/crosstool-NG/builds/xtensa-esp32-elf/bin
IDFPATH=/home/robert/esp/esp-idf

makefile=
ESPIDF = $(HOME)/esp/esp-idf/
PORT = /dev/ttyUSB0
FLASH_MODE = dio
FLASH_SIZE = 4MB
#CROSS_COMPILE = xtensa-esp32-elf-

include Makefile

Any ideas on what's causing the error?

thx robert

`robert@dg41rq:~/esp/micropython-esp32/esp32$ make
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
** WARNING **
The git hash of ESP IDF does not match the supported version
The build may complete and the firmware may work but it is not guaranteed
ESP IDF path:       /home/robert/esp/esp-idf/
Current git hash:   fa7d53e700c6432aff6dbbe41dbab3bdd366da6f
Supported git hash: 4ec2abbf23084ac060679e4136fa222a2d0ab0e8
mkdir -p build/bootloader//home/robert/esp/esp-idf//components/bootloader/src/main/
mkdir -p build/bootloader//home/robert/esp/esp-idf//components/bootloader_support/src/
mkdir -p build/bootloader//home/robert/esp/esp-idf//components/log/
mkdir -p build/bootloader//home/robert/esp/esp-idf//components/micro-ecc/micro-ecc/
mkdir -p build/bootloader//home/robert/esp/esp-idf//components/soc/esp32/
mkdir -p build/bootloader//home/robert/esp/esp-idf//components/spi_flash/
CC /home/robert/esp/esp-idf//components/bootloader_support/src/bootloader_flash.c
CC /home/robert/esp/esp-idf//components/bootloader_support/src/bootloader_random.c
CC /home/robert/esp/esp-idf//components/bootloader_support/src/secure_boot_signatures.c
CC /home/robert/esp/esp-idf//components/bootloader_support/src/secure_boot.c
CC /home/robert/esp/esp-idf//components/bootloader_support/src/esp_image_format.c
CC /home/robert/esp/esp-idf//components/bootloader_support/src/flash_encrypt.c
CC /home/robert/esp/esp-idf//components/bootloader_support/src/flash_partitions.c
CC /home/robert/esp/esp-idf//components/log/log.c
CC /home/robert/esp/esp-idf//components/spi_flash/spi_flash_rom_patch.c
CC /home/robert/esp/esp-idf//components/soc/esp32/rtc_clk.c
CC /home/robert/esp/esp-idf//components/soc/esp32/rtc_time.c
CC /home/robert/esp/esp-idf//components/micro-ecc/micro-ecc/uECC.c
LINK build/bootloader.elf
xtensa-esp32-elf-gcc: error: build/bootloader//home/robert/esp/esp-idf//components/bootloader/src/main/bootloader_start.o: No such file or directory
Makefile:687: recipe for target 'build/bootloader.elf' failed
make: *** [build/bootloader.elf] Error 1`

robert-hh, 2017-07-22:
Hello Robert, I stumbled over this error too this morning,. The reason is a mismatch between the supported esp-idf version and the actual one. Go to your esp-if directory and issue the commands:
```
git checkout 4ec2abbf23084ac060679e4136fa222a2d0ab0e8
git submodule update --init
```
That will set the esp-idf to the supported version.

TrifidSagitarius, 2017-07-22:
Robert,

Thanks for the heads up, that did the trick.
Looks like I'll be spending a lot of time in this forums :)

robert,

closed 2017-07-23 by TrifidSagitarius

sravanth005, 2019-03-11:
Serial port /dev/ttyUSB1
Traceback (most recent call last):
  File "/home/admin-unizen/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 2959, in <module>
    _main()
  File "/home/admin-unizen/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 2952, in _main
    main()
  File "/home/admin-unizen/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 2652, in main
    esp = chip_class(each_port, initial_baud, args.trace)
  File "/home/admin-unizen/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 222, in __init__
    self._port = serial.serial_for_url(port)
  File "/usr/lib/python2.7/dist-packages/serial/__init__.py", line 85, in serial_for_url
    instance.open()
  File "/usr/lib/python2.7/dist-packages/serial/serialposix.py", line 294, in open
    raise SerialException(msg.errno, "could not open port %s: %s" % (self._port, msg))
serial.serialutil.SerialException: [Errno 2] could not open port /dev/ttyUSB1: [Errno 2] No such file or directory: '/dev/ttyUSB1'
/home/admin-unizen/esp/esp-idf/components/esptool_py/Makefile.projbuild:64: recipe for target 'flash' failed
make: *** [flash] Error 1

**i successfully installed toolchain ,My setup here is ubuntu 16.04 and I have xtensa and espressif installed. but at last when i flash,i end up with this error so how to move on with this error if anyone knew it please send the commands so that i can follow accordingly.**

dpgeorge, 2019-03-11:
@sravanth005 you need to select/use the correct serial port for your particular board.  Best to get help for these kinds of questions at https://forum.micropython.org

sravanth005, 2019-03-12:
**i work in a project with esp32-cam (https://robotzero.one/esp32-camera-module/#comment-3073) refer the link. 
i followed the link above though it scripted foe windows os for some commands i went through (https://docs.espressif.com/projects/esp-idf/en/latest/get-started/linux-setup.html#) link and i installed.but after flashing at the end i get error like this** 

Serial port /dev/ttyUSB1
Traceback (most recent call last):
File "/home/admin-unizen/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 2959, in
_main()
File "/home/admin-unizen/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 2952, in _main
main()
File "/home/admin-unizen/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 2652, in main
esp = chip_class(each_port, initial_baud, args.trace)
File "/home/admin-unizen/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 222, in init
self._port = serial.serial_for_url(port)
File "/usr/lib/python2.7/dist-packages/serial/init.py", line 85, in serial_for_url
instance.open()
File "/usr/lib/python2.7/dist-packages/serial/serialposix.py", line 294, in open
raise SerialException(msg.errno, "could not open port %s: %s" % (self._port, msg))
serial.serialutil.SerialException: [Errno 2] could not open port /dev/ttyUSB1: [Errno 2] No such file or directory: '/dev/ttyUSB1'
/home/admin-unizen/esp/esp-idf/components/esptool_py/Makefile.projbuild:64: recipe for target 'flash' failed
make: *** [flash] Error 1

**This is my path which i set:
export PATH=$PATH:/home/admin-unizen/esp32/xtensa-esp32-elf/bin
export IDF_PATH=/home/admin-unizen/esp/esp-idf

i dont know how to solve.can you explain me fully why this happening and will you say the commands from first to last to solve this issue**

sravanth005, 2019-03-12:
(https://robotzero.one/esp32-camera-module/#comment-3073) refer the link.
im currently trying to run this unit ,after i do 'make monitor' the terminal window is keep on running without any pause and ip address is not coming can any one explain how to check that ip address and where i should type it to get that GUI window  ,i even tried in a browser but its not coming it takes more and displays connection timed out.

sravanth005, 2019-03-13:
I (776) cpu_start: App cpu up.
I (1279) spiram: SPI SRAM memory test OK
I (1280) heap_init: Initializing. RAM available for dynamic allocation:
I (1280) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (1286) heap_init: At 3FFBC470 len 00023B90 (142 KiB): DRAM
I (1293) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1299) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1306) heap_init: At 400977EC len 00008814 (34 KiB): IRAM
I (1312) cpu_start: Pro cpu start user code
I (1317) spiram: Adding pool of 4096K of external SPI memory to heap allocator
I (218) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (267) wifi: wifi driver task: 3ffc3ea0, prio:23, stack:3584, core=0
I (267) wifi: wifi firmware version: 955b7af
I (267) wifi: config NVS flash: enabled
I (277) wifi: config nano formating: disabled
I (277) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (287) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (307) wifi: Init dynamic tx buffer num: 32
I (307) wifi: Init data frame dynamic rx buffer num: 32
I (307) wifi: Init management frame dynamic rx buffer num: 32
I (317) wifi: Init management short buffer num: 32
I (317) wifi: Init static rx buffer size: 1600
I (327) wifi: Init static rx buffer num: 10
I (327) wifi: Init dynamic rx buffer num: 32
I (407) camera wifi: wifi_init_softap finished.SSID:ESP32-Camera password:
I (417) camera wifi: wifi_init_sta finished.
I (417) camera wifi: connect to ap SSID:Unizen_gmesh1 password:Gmesh@2018

Brownout detector was triggered

ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:6816
ho 0 tail 12 room 4
load:0x40078000,len:12116
load:0x40080400,len:7400
entry 0x40080780
I (31) boot: ESP-IDF v3.3-beta2-8-g89ae590-dirty 2nd stage bootloader
I (31) boot: compile time 18:30:26
I (32) boot: Enabling RNG early entropy source...
I (37) qio_mode: Enabling default flash chip QIO
I (43) boot: SPI Speed      : 80MHz
I (47) boot: SPI Mode       : QIO
I (51) boot: SPI Flash Size : 4MB
I (55) boot: Partition Table:
I (58) boot: ## Label            Usage          Type ST Offset   Length
I (66) boot:  0 factory          factory app      00 00 00010000 00300000
I (73) boot:  1 nvs              WiFi data        01 02 00310000 00004000
I (81) boot: End of partition table
I (85) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x17fb0c (1571596) map
I (509) esp_image: segment 1: paddr=0x0018fb34 vaddr=0x3ffb0000 size=0x004dc (  1244) load
I (510) esp_image: segment 2: paddr=0x00190018 vaddr=0x400d0018 size=0x8c0d0 (573648) map
0x400d0018: _stext at ??:?

I (667) esp_image: segment 3: paddr=0x0021c0f0 vaddr=0x3ffb04dc size=0x02dbc ( 11708) load
I (671) esp_image: segment 4: paddr=0x0021eeb4 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at /home/admin-unizen/esp/esp-idf/components/freertos/xtensa_vectors.S:1779

I (674) esp_image: segment 5: paddr=0x0021f2bc vaddr=0x40080400 size=0x173ec ( 95212) load
I (728) boot: Loaded app from partition at offset 0x10000
I (728) boot: Disabling RNG early entropy source...
I (729) spiram: Found 64MBit SPI RAM device
I (733) spiram: SPI RAM mode: flash 80m sram 80m
I (738) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (745) cpu_start: Pro cpu up.
I (749) cpu_start: Application information:
I (754) cpu_start: Project name:     camera_web_server
I (760) cpu_start: App version:      v0.9.1-12-g058c427
I (766) cpu_start: Compile time:     Mar 12 2019 18:30:28
I (772) cpu_start: ELF file SHA256:  2d7ce3724cb3dee7...
I (778) cpu_start: ESP-IDF:          v3.3-beta2-8-g89ae590-dirty
I (785) cpu_start: Starting app cpu, entry point is 0x4008145c
0x4008145c: call_start_cpu1 at /home/admin-unizen/esp/esp-idf/components/esp32/cpu_start.c:267

I (776) cpu_start: App cpu up.
I (1279) spiram: SPI SRAM memory test OK
I (1280) heap_init: Initializing. RAM available for dynamic allocation:
I (1280) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (1286) heap_init: At 3FFBC470 len 00023B90 (142 KiB): DRAM
I (1293) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1299) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1306) heap_init: At 400977EC len 00008814 (34 KiB): IRAM
I (1312) cpu_start: Pro cpu start user code
I (1317) spiram: Adding pool of 4096K of external SPI memory to heap allocator
I (218) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (267) wifi: wifi driver task: 3ffc3f24, prio:23, stack:3584, core=0
I (267) wifi: wifi firmware version: 955b7af
I (267) wifi: config NVS flash: enabled
I (277) wifi: config nano formating: disabled
I (277) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (287) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (307) wifi: Init dynamic tx buffer num: 32
I (307) wifi: Init data frame dynamic rx buffer num: 32
I (307) wifi: Init management frame dynamic rx buffer num: 32
I (317) wifi: Init management short buffer num: 32
I (317) wifi: Init static rx buffer size: 1600
I (327) wifi: Init static rx buffer num: 10
I (327) wifi: Init dynamic rx buffer num: 32
I (337) camera wifi: wifi_init_softap finished.SSID:ESP32-Camera password:
I (397) camera wifi: wifi_init_sta finished.
I (397) camera wifi: connect to ap SSID:Unizen_gmesh1 password:Gmesh@2018

Brownout detector was triggered

ets Jun  8 2016 00:22:57

Whats the problem i can see that "Brownout detector was triggered" plenty of times,how solve this issue please comment

robert-hh, 2019-03-13:
Brownout is a type of power failure. Check the power supply and the cable.

sravanth005, 2019-03-14:
thanks @robert-hh  i checked the power supply and i came to know it was not sufficient and i fixed it now its working

sravanth005, 2019-03-14:
Hi all,
I am try to work with ESP32-CAM (https://robotzero.one/esp32-camera-module/#comment-3073) i can able to take a still with the camera but face detection and face reorganization is not happening, Do any one know for what reason it is not happening.please help me out to solve this issue.

thanks in advance

gouravjha7, 2021-04-15:
Getting error while building program

> Executing task: cmake -G Ninja .. <

-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32
-- Checking Python dependencies...
Python requirements from C:\Users\Gourav\esp\esp-idf\requirements.txt are satisfied.
-- Could NOT find Perl (missing: PERL_EXECUTABLE) 
-- App "hello-world" version: 1
-- Adding linker script C:/Users/Gourav/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-time.ld
-- Adding linker script C:/Users/Gourav/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script C:/Users/Gourav/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld     
-- Adding linker script C:/Users/Gourav/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script C:/Users/Gourav/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script C:/Users/Gourav/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script C:/Users/Gourav/Documents/ESP32 Codes/hello_world/build/esp-idf/esp32/esp32_out.ld
-- Adding linker script C:/Users/Gourav/esp/esp-idf/components/esp32/ld/esp32.project.ld.in
-- Adding linker script C:/Users/Gourav/esp/esp-idf/components/esp32/ld/esp32.peripherals.ld
-- Components: app_trace app_update asio bootloader bootloader_support bt cbor coap console cxx driver efuse esp-tls esp32 esp_adc_cal esp_common 
esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_ipc esp_local_ctrl esp_netif esp_ringbuf 
esp_rom esp_serial_slave_link esp_system esp_timer esp_websocket_client esp_wifi espcoredump esptool_py expat fatfs freemodbus freertos heap idf_test jsmn json libsodium log lwip mbedtls mdns mqtt newlib nghttp nvs_flash openssl partition_table perfmon protobuf-c protocomm pthread sdmmc soc 
spi_flash spiffs tcp_transport tcpip_adapter tinyusb ulp unity vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: C:/Users/Gourav/esp/esp-idf/components/app_trace C:/Users/Gourav/esp/esp-idf/components/app_update C:/Users/Gourav/esp/esp-idf/components/asio C:/Users/Gourav/esp/esp-idf/components/bootloader C:/Users/Gourav/esp/esp-idf/components/bootloader_support C:/Users/Gourav/esp/esp-idf/components/bt C:/Users/Gourav/esp/esp-idf/components/cbor C:/Users/Gourav/esp/esp-idf/components/coap C:/Users/Gourav/esp/esp-idf/components/console C:/Users/Gourav/esp/esp-idf/components/cxx C:/Users/Gourav/esp/esp-idf/components/driver C:/Users/Gourav/esp/esp-idf/components/efuse C:/Users/Gourav/esp/esp-idf/components/esp-tls C:/Users/Gourav/esp/esp-idf/components/esp32 C:/Users/Gourav/esp/esp-idf/components/esp_adc_cal C:/Users/Gourav/esp/esp-idf/components/esp_common C:/Users/Gourav/esp/esp-idf/components/esp_eth C:/Users/Gourav/esp/esp-idf/components/esp_event C:/Users/Gourav/esp/esp-idf/components/esp_gdbstub C:/Users/Gourav/esp/esp-idf/components/esp_hid C:/Users/Gourav/esp/esp-idf/components/esp_http_client C:/Users/Gourav/esp/esp-idf/components/esp_http_server C:/Users/Gourav/esp/esp-idf/components/esp_https_ota C:/Users/Gourav/esp/esp-idf/components/esp_https_server C:/Users/Gourav/esp/esp-idf/components/esp_ipc C:/Users/Gourav/esp/esp-idf/components/esp_local_ctrl C:/Users/Gourav/esp/esp-idf/components/esp_netif C:/Users/Gourav/esp/esp-idf/components/esp_ringbuf C:/Users/Gourav/esp/esp-idf/components/esp_rom C:/Users/Gourav/esp/esp-idf/components/esp_serial_slave_link C:/Users/Gourav/esp/esp-idf/components/esp_system C:/Users/Gourav/esp/esp-idf/components/esp_timer C:/Users/Gourav/esp/esp-idf/components/esp_websocket_client C:/Users/Gourav/esp/esp-idf/components/esp_wifi C:/Users/Gourav/esp/esp-idf/components/espcoredump C:/Users/Gourav/esp/esp-idf/components/esptool_py C:/Users/Gourav/esp/esp-idf/components/expat C:/Users/Gourav/esp/esp-idf/components/fatfs C:/Users/Gourav/esp/esp-idf/components/freemodbus C:/Users/Gourav/esp/esp-idf/components/freertos C:/Users/Gourav/esp/esp-idf/components/heap C:/Users/Gourav/esp/esp-idf/components/idf_test C:/Users/Gourav/esp/esp-idf/components/jsmn C:/Users/Gourav/esp/esp-idf/components/json C:/Users/Gourav/esp/esp-idf/components/libsodium C:/Users/Gourav/esp/esp-idf/components/log C:/Users/Gourav/esp/esp-idf/components/lwip C:/Users/Gourav/esp/esp-idf/components/mbedtls C:/Users/Gourav/esp/esp-idf/components/mdns C:/Users/Gourav/esp/esp-idf/components/mqtt C:/Users/Gourav/esp/esp-idf/components/newlib C:/Users/Gourav/esp/esp-idf/components/nghttp C:/Users/Gourav/esp/esp-idf/components/nvs_flash C:/Users/Gourav/esp/esp-idf/components/openssl C:/Users/Gourav/esp/esp-idf/components/partition_table C:/Users/Gourav/esp/esp-idf/components/perfmon C:/Users/Gourav/esp/esp-idf/components/protobuf-c C:/Users/Gourav/esp/esp-idf/components/protocomm C:/Users/Gourav/esp/esp-idf/components/pthread C:/Users/Gourav/esp/esp-idf/components/sdmmc C:/Users/Gourav/esp/esp-idf/components/soc C:/Users/Gourav/esp/esp-idf/components/spi_flash C:/Users/Gourav/esp/esp-idf/components/spiffs C:/Users/Gourav/esp/esp-idf/components/tcp_transport C:/Users/Gourav/esp/esp-idf/components/tcpip_adapter C:/Users/Gourav/esp/esp-idf/components/tinyusb C:/Users/Gourav/esp/esp-idf/components/ulp C:/Users/Gourav/esp/esp-idf/components/unity C:/Users/Gourav/esp/esp-idf/components/vfs C:/Users/Gourav/esp/esp-idf/components/wear_levelling C:/Users/Gourav/esp/esp [... cut ...] T bootloader.ld  -T bootloader.rom.ld && cd ."
xtensa-esp32-elf-gcc.exe: error: Codes/hello_world/build/bootloader/bootloader.map: No such file or directory
ninja: build stopped: subcommand failed.
[3/355] Linking CXX static library esp-idf\esp-tls\libesp-tls.a
ninja: build stopped: subcommand failed.
The terminal process "C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe -Command cmake --build ." terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

marcrobm, 2021-05-03:
> ng error while building p

Did you find a way to solve that issue, and would like to let us know?