micropython/micropython-esp32 #137: Ssl possible memory leak on server side

LupascuAndrei, 2017-07-24:
https://github.com/micropython/micropython-esp32/blob/esp32/extmod/modussl_mbedtls.c#L231-L236

Shouldn't the <pkey> also be freed there?  Esp32 crashes once every 100-150 ssl handshakes when run as ssl server mode; i *think* that clearing the pkey fixed it for me
```C
    mbedtls_x509_crt_free(&self->cert);
    mbedtls_ssl_free(&self->ssl);
    mbedtls_ssl_config_free(&self->conf);
    mbedtls_ctr_drbg_free(&self->ctr_drbg);
    mbedtls_entropy_free(&self->entropy);
    mbedtls_pk_free(&self->pkey); // this
```

dpgeorge, 2017-07-25:
Thanks for the report, it does look like a bug with the freeing of resources.  I fixed it upstream at the patch should make its way here soon (I managed to get 1000+ successful connections in server mode with the patch).

> Also, regarding the same file, why is null_entropy_func used?

See #122 for a discussion about that.

dpgeorge, 2017-07-26:
Resource freeing is now merged, see 0893b273b9471acb02510820f61753501dc04219

closed 2017-07-26 by dpgeorge