micropython/micropython-esp32 #138: FEATURE mbedtls certificate generation

LupascuAndrei, 2017-07-25:
Hello ! I've implemented rsa key+(self signed) certificate generation into micropython-esp32 using mbedtls. 
I'm asking if the idea to be able to generate a new key/cert pair would be considered useful enough to be integrated into this repo.

My first go at it can be found here: 

https://github.com/LupascuAndrei/dump/tree/master/micropython_mbedtls_key_generation_esp32

An example of using it from python:

```python 
import machine
cert = machine.Certificate()
pair = cert.generate()
print(pair[0])
print(pair[1])
print('@@@@@@@@@@@@')
pair = cert.generate()
print(pair[0])
print(pair[1])
```

prints the following: 
```
-----BEGIN RSA PRIVATE KEY-----
MIICXAIBAAKBgQCGlVUhhsGa1nRxNrFvvkp3pElII8gC601nDBC+AS6Scfdg3fV1
in7H3zlnjzaILu2zXXYwKvbAJVDw6gVFlYsyS+B0m1+TPCzLrzdVrd67yZBtbeky
qbbhB/4dR0BMckWA/xBfWwp537kTiG20y+YldIMELiWeokHsPjYEHt8+SQIDAQAB
AoGAbSgDw0yMDOzxxr2Ijsx1k7JxKYyz9gIpZnH3xlQpFOhQr4V3R92L1MrtwVv7
69q8OqWMbab7OX9KtgvpOLYBDhd18K3R7rG5JSIqtYo/cdpJhVHwr+Ph8PLq/LS4
m8Q1+DqwuKA71NPBPFPVClFS8d8q2OB5Qgt6EYmC/6zhg4ECQQDlWMIFceJPN0fD
FnHYMz3ybDkOLDlIAc53mYKAG6LEap92UQMVw8xJo8OQLUtgdRJLNljfXjyMrJmB
+IJIm8ZxAkEAljlKbwpSBC2iHO+FgbOBTLXZq4nvE8iXzppYMR2A9HwEHY0H1vdB
YoWQ0++42ZWw1JSsrcQmJ2vtv5qoy4XRWQJBAJfW/JVYQamKgtNRP8wQmNmv9tJA
Syg2MfiSxIiGQ2vg33QYzH8TY2P5rRb+NeJFf/gVoh84J8RG3VbkLXVe1LECQArW
Z0BLhk8225Mq1H7e7F0UYXe2o8DRn+IP0Hz6m+sHZ2SaYA5qGBV7fU9M5BEOc4ti
jnKazAXcC3LfPX4xUUECQD+qqSPjITeAV+wvgxE6921RlzrpdkkOis8B4oO9XvrE
ek37S+eTLjnYbJ4+zVgZ4Pt5wvExMHymvH7BiED23q4=
-----END RSA PRIVATE KEY-----

-----BEGIN CERTIFICATE-----
MIICHTCCAYagAwIBAgIBATANBgkqhkiG9w0BAQsFADAtMQswCQYDVQQDEwJDQTER
MA8GA1UEChMIbWJlZCBUTFMxCzAJBgNVBAYTAlVLMB4XDTAxMDEwMTAwMDAwMFoX
DTMwMTIzMTIzNTk1OVowLTELMAkGA1UEAxMCQ0ExETAPBgNVBAoTCG1iZWQgVExT
MQswCQYDVQQGEwJVSzCBnzANBgkqhkiG9w0BAQEFAAOBjQAwgYkCgYEAhpVVIYbB
mtZ0cTaxb75Kd6RJSCPIAutNZwwQvgEuknH3YN31dYp+x985Z482iC7ts112MCr2
wCVQ8OoFRZWLMkvgdJtfkzwsy683Va3eu8mQbW3pMqm24Qf+HUdATHJFgP8QX1sK
ed+5E4httMvmJXSDBC4lnqJB7D42BB7fPkkCAwEAAaNNMEswCQYDVR0TBAIwADAd
BgNVHQ4EFgQUY8igoWSKhdTVnqQKHa+AeHSNQpIwHwYDVR0jBBgwFoAUY8igoWSK
hdTVnqQKHa+AeHSNQpIwDQYJKoZIhvcNAQELBQADgYEACQLq32G/nOC0zBAG9tS5
QPlKnRzs8g8AxMDwk382pm8uGg5xZNgxnBrNVyt+VErvtk6zLo979Bz09GWt9rwB
kOkIWnzVxDaQHF/88QAFU1XcA7swglhMWIUOex93J8NQeDh+qYk2f6haqtd/8u++
XxGmGa2BvgnG1J3PY/wmdEQ=
-----END CERTIFICATE-----

@@@@@@@@
-----BEGIN RSA PRIVATE KEY-----
MIICXQIBAAKBgQCCW9s2L36MpmVQmOIaMppr8q1pmTaXaS1agh+ZkpRyUReH9FSr
tFTbZ7uNifP6TMtLNdzipWeqiRkGF2R2V8BnpcgWHdkqiVvh4cUIdL8YGNdk+LFC
4rFwuziIlhPoZgX5jASskNnwMQZLZF1Nq95RoUtvAKDFRmnPvGNZBx3PwQIDAQAB
AoGAHQA6gI+IY9oQ6CFbC+D+Ic3Xboq7bmvjvyL7zLqjc9s+lOxyedCgoD8vpolu
Ud0LzSuqfHG34mSlwIkrvHhBQXoCU8MpxHOfQI9xHZKMuDvJPY3Pq2y/KAQTpSfY
YS7tMa1OoVNFOCZM71QRf0m38Z/Tul8/U8bw7NJWkiNpXtECQQDqBsL6eKJlVsOc
s+GJVunzQ8UeGh878VdYviooPwpnCMeYS33GOE4kIXUDYy+0wtHYTjNfjA194Pos
dk0kyj7XAkEAjplCv2iLjseXO/o8+ol9/jheM9XV91Uw2yUDjmSL1dxiuF71BsbL
6vPEu5A3eUNVOlJjHkbhrDfcCV6s5N0LJwJBAIKTyUDmBH6Uvcp2l7YA18KTIq4c
FydkDr+GQ5F3Mz+wJ2XzYTczfW4teNOuCRVLyrhtUb7TL0f4Or6KN5r6ebsCQEYx
Yw3dZHUQAn3fnAyniYGONbDGXeC16qq0+Z8qFztalK208GA5kz9AZ2L8hpIGdvmS
fH/23Ddt974VAH5X4JkCQQCmijYTB/7WAjAftZz6pc+P1wocMW7AuKRaOWU6NRPe
vmfeXpNoHLbFV+LLvuiE558nvBvyy3L9cbocu/YkF2Nd
-----END RSA PRIVATE KEY-----

-----BEGIN CERTIFICATE-----
MIICHTCCAYagAwIBAgIBATANBgkqhkiG9w0BAQsFADAtMQswCQYDVQQDEwJDQTER
MA8GA1UEChMIbWJlZCBUTFMxCzAJBgNVBAYTAlVLMB4XDTAxMDEwMTAwMDAwMFoX
DTMwMTIzMTIzNTk1OVowLTELMAkGA1UEAxMCQ0ExETAPBgNVBAoTCG1iZWQgVExT
MQswCQYDVQQGEwJVSzCBnzANBgkqhkiG9w0BAQEFAAOBjQAwgYkCgYEAglvbNi9+
jKZlUJjiGjKaa/KtaZk2l2ktWoIfmZKUclEXh/RUq7RU22e7jYnz+kzLSzXc4qVn
qokZBhdkdlfAZ6XIFh3ZKolb4eHFCHS/GBjXZPixQuKxcLs4iJYT6GYF+YwErJDZ
8DEGS2RdTaveUaFLbwCgxUZpz7xjWQcdz8ECAwEAAaNNMEswCQYDVR0TBAIwADAd
BgNVHQ4EFgQUi4/zyevLlww7LfA7N8cLYkKJV9YwHwYDVR0jBBgwFoAUi4/zyevL
lww7LfA7N8cLYkKJV9YwDQYJKoZIhvcNAQELBQADgYEALRhuntXcP8nOjddS0RRA
Ko5M8pCSTy/EpBISMRGToyzzklPIMZWD7wPauKwGKjb4oevONDFhuwMvkm1cb5o2
ZdwZc8racTE6yhc2sx3buAOiOx5p9eEqgoPac7e5wAYOEbOy1SWrzQEGKrq6XGrQ
aKPifw7fBTDO1hN3AalbVW0=
-----END CERTIFICATE-----
```

It's stable, I *think* there are no memory leaks ( 1300 pairs generated so far in a loop, still going, will return with better results ).

I think there aren't many cases where this is needed and i'm asking if something like this would be desired in this project.

Btw, thank you a lot for the esp32 port, it's working great so far :D

dpgeorge, 2017-07-26:
Key/cert generation is the kind of thing that's best done offline on a machine that has the resources to do it.  But feel free to provide examples/discussion as to why it's beneficial to do in on the board (maybe it's useful to generate keys, but I doubt it's useful to regenerate certificates).

dpgeorge, 2019-10-22:
Closing.  Feel free to restart the discussion at https://forum.micropython.org

closed 2019-10-22 by dpgeorge