micropython/micropython-esp32 #143: AR/VR Rig

DVSUHZ-g, 2017-07-28:
I'm making an augmented reality headset and have been doing research for +8months (lost 90% of it a week ago). 

The rough sketch is this:
[image]

Eventually working toward using these respos extensively:
[image]

Feedback welcome.

MrSurly, 2017-07-28:
Hi @DVSUHZ-g,

Are you having any specific problems with MicroPython?  Issues are usually places to post a problem with the code, or perhaps a proposed change.

DVSUHZ-g, 2017-07-28:
As stated in:
https://youtu.be/-MrqCmq3Z5k
*Oi! I wish it could do this.*