micropython/micropython-esp32 #145: import network causes reboot

rdagger, 2017-07-30:
Just installed today's firmware build: esp32-20170730-v1.9.1-391-g06168c92.bin
The command `import network` is now causing the ESP32 to reboot.
It's also causing problems with the umqtt.simple library.

rdagger, 2017-07-30:
Here's the error message: 

>>> import network
I (132277) wifi: wifi firmware version: f092575
I (132277) wifi: config NVS flash: enabled
I (132277) wifi: config nano formating: disabled
I (132277) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (132287) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (132307) wifi: Init dynamic tx buffer num: 32
I (132307) wifi: Init dynamic rx buffer num: 64
I (132307) wifi: wifi driver task: 3ffd91e8, prio:23, stack:4096
I (132307) wifi: Init static rx buffer num: 10
I (132317) wifi: Init dynamic rx buffer num: 0
I (132317) wifi: Init rx ampdu len mblock:7
I (132327) wifi: Init lldesc rx ampdu entry mblock:4
I (132327) wifi: wifi power manager task: 0x3ffde4d4 priautoconnect: /dev/ttyUSB0 action: remove
USB Serial device '/dev/ttyUSB0' disconnected

MrSurly, 2017-07-30:
I've had this problem before. The fact that the USB disconnected is a pretty clear sign that your power is browning out.  This happens because WiFi draws considerably more current when it fires up, and if you have a high resistance on the power wires, this can happen.

Try using use a shorter USB cable (seriously).

dpgeorge, 2017-07-31:
The latest code works for me.  The ESP IDF recently changed the way it handles brown-out detection so that's further evidence it could be related to power.

rdagger, 2017-07-31:
Thanks, you were right!  I added an additional 5V power line and it's working now.

dpgeorge, 2017-07-31:
You can try making the CPU run slower to use less power: it's a compile time option, grep for CONFIG_ESP32_DEFAULT_CPU_FREQ

closed 2017-07-31 by dpgeorge