micropython/micropython-esp32 #147: Current ESP32 image (esp32-20170731...) boot failure

heuckeg, 2017-07-31:
The header (the first 10 bytes of the bin bile) of this file (and previous bin's) is:
E9 04 02 20 xx xx xx xx EE 00

The ninth byte is wrong. It's a marker for a encoded bin file. I changed this byte manually to 00 that it looks like this:
E9 04 02 20 xx xx xx xx 00 00

Then micro phython was starting up on a esp32 wroom based board.

I don't know where to change in the source code of micropython.

Grettings,
Gunter

dpgeorge, 2017-07-31:
Please note that images must now be flashed at the location 0x1000.

nickzoic, 2017-08-03:
Hi Damien, if the flash has been corrupted can the data at 0x0 be restored somehow?

dpgeorge, 2017-08-03:
> if the flash has been corrupted can the data at 0x0 be restored somehow?

You can read the flash using `esptool.py read_flash` and use that to make a backup.  Otherwise best thing to do is just erase everything and start again.

closed 2019-01-16 by dpgeorge