micropython/micropython-esp32 #153: from network import Bluetooth Documentation/Examples?

RicardoDanielSimoes, 2017-08-06:
Hello,

We're currently changing Hardware from WiPy 2.0 to ESP32 to test some things out, and after putting the uPy Firmware on it and testing it, it works. Current problem is that due to several changes in how BLE is handled codewise from WiPy to ESP32 uPY, we cannot seem to get the same stuff running.

robert-hh, 2017-08-06:
I that is the only issue, you can run the Wipy 2 code on almost any esp32 board or module, like the ESP32 Thing, Wemos LoLin32, Huzzah ESP32. Besides that, I expect this ESP32 port to mature and get as robust as the ones for PyBoard and ESP8266 by having a smaller footprint, more available heap space and a friendly license model.

RicardoDanielSimoes, 2017-08-06:
We have gotten ourselfes the ESP32 DevKitC as we are in a hurry and had to take what was around. We're using uPy IDE to push the Firmware onto the ESP32 (Works flawlessly). Sadly the WiPy 2 Code does not work on the current Micropython FW, as several stuff appears to be different. 

```
#Importe
from network import Bluetooth
import binascii 
import dataStore #Import DataStore

class bluetooth:
    data = dataStore.DataStore ()
    bluetooth = Bluetooth()
    bluetooth.init()
    
    def __init__(self,data,):
        #Init Bluetooth and enable Scanning
        self.data = data 
        if not self.bluetooth.isscanning():
            self.bluetooth.start_scan(-1)
      
    
    def scan(self):
        adv =self.bluetooth.get_adv()       
        try: 
            #look if the receved mac is in the data store
             for x in range(0, (self.data.getAnzahlTokens())):                
                if str(binascii.hexlify(adv.mac)) == str(self.data.getnextMac()): 
                        #lay down the RSSI value to the DataStore
                        print('*********************************************')
                        print("Geraete Mac Adresse = {}".format(binascii.hexlify(adv.mac)))
                        print("Geraete Sendestaerke = {}".format(adv.rssi)) 
                        data_stream= str(adv.data)
                        self.data.setRssi(binascii.hexlify(adv.mac), adv.rssi)
                        self.data.setDaten(binascii.hexlify(adv.mac), data_stream.replace("\\", ";"))
                        break
        except:
            pass

```

This, for example, gives us something strange about not defining a MAC address, which we cannot seem to get solved.

robert-hh, 2017-08-06:
I was talking about the WiPy 2 firmware itself, not your app.

RicardoDanielSimoes, 2017-08-06:
Yeah, but we're trying out some different Firmware as we're having some problems with the Bluetooth on the WiPy 2.0. We wanted to see if the same happens with a normal ESP32 board, so that we can communicate with our Customer.

MrSurly, 2017-08-06:
@RicardoDanielSimoes 

I assume you're referring to https://github.com/micropython/micropython-esp32/pull/86?

I'm the author of that code.  It's implementation was done without referencing the PyCom code, or the API it exposes.  This was intentional -- the PyCom license isn't compatible with μPy.  The good news is that it supports descriptors, and AFAIK the PyCom version does not.

In any case, you'll see that there's some discussion in that thread WRT the final form the the API itself.  It's not part of the official ESP32 port yet, and even then, the entire ESP32 port is considered experimental and subject to change.

RicardoDanielSimoes, 2017-08-08:
I see.

Thing is before we subject ourselfs to start lerning C++ and programm the whole stuff on ESP32 natively in less than a week, we wanted to try to implement the same stuff on ESP32 with uPy....
The PyCom one is pretty interessting but is plagued by unstabillity, which is a shame as the rest works flawlessly.

dpgeorge, 2019-09-25:
BLE support for esp32 is being implemented upstream via the following PR: micropython/micropython#5051

closed 2019-09-25 by dpgeorge