micropython/micropython-esp32 #154: Support for ULP processor

nickzoic, 2017-08-07:
There's no support for the ULP processor yet.
API docs: https://github.com/espressif/esp-idf/blob/master/docs/api-guides/ulp.rst

nickzoic, 2017-08-07:
The CPU is different to the main CPU cores and is unlikely to ever run Python bytecode, but I was thinking it might be programmable from Python as a kind of a DSL.

eduvik, 2017-08-07:
I'd suggest coming up with a bunch of use cases for it, then determining how much to generalise from that. If there's a small number, perhaps just implement some python hooks to those cases (might also be useful to do this for the most common cases, implemented in a DSL). 

I'll get started with some use cases:

- Wake up main CPU based on something simple. This could be a GPIO change, or a timer.
- Wake up main CPU based on something complex. This could be a sensor value where the sensor speaks a complex protocol (looks like I2C is supported) that the ULP would have to interrogate itself. This might require a driver set that the ULP can speak with
- Low power sensor logging. This will depend on how much access this has to memory or storage. 
- Combinations of the above - e.g. log sensor value every 10 minutes, wake up main CPU every 2 hours (to send to internet), and wake up if value drops below X

mattytrentini, 2017-08-12:
I'm interested in that 3rd use case; logging data from a low-power sensor.

The ULP has access to the 8K of RTC memory (see the [ULP coprocessor instruction set](http://esp-idf.readthedocs.io/en/latest/api-guides/ulp_instruction_set.html)). So the idea would be to put the main processor to sleep, use the ULP to read the sensor periodically, fill up the 8K and wake the main processor to offload the 8K to persistent storage. Should use only a tiny amount of power.

I'm not sure what peripherals the ULP has access to exactly but that document above indicates it can work with at least I2C - which covers most of the sensors I typically deal with...

MrSurly, 2017-08-12:
@nickzoic 

In the deepsleep [PR](https://github.com/micropython/micropython-esp32/pull/85), I added a stubby for RTC memory -- the IDF wasn't ready then, either.  I do seem to remember some examples, though?

```c
STATIC mp_obj_t machine_rtc_memory(mp_uint_t n_args, const mp_obj_t *args) {

    if (n_args == 1) {
        // read RTC memory

        // FIXME; need to update IDF

        return mp_const_none;
    } else {
        // write RTC memory

        // FIXME; need to update IDF

        return mp_const_none;
    }
}
```

nickzoic, 2017-08-13:
So I was thinking about this earlier: the ULP just needs a little bytestring of machine code sent to it to configure it, so as a first step having an `esp.ulp.load(bytes)` would work I think.  Then the question becomes: from where do we get the bytes?

What I'd *really* like is a python bytecode -> ULP assembly translator but that might be a bit ambitious.

There's some precendent here in the Thumb assembler https://docs.micropython.org/en/latest/pyboard/reference/asm_thumb2_hints_tips.html and maybe we could do something similar for the ULP assembler. I'm not a fan of the syntax though.

I was messing around with `with` syntax, so you could create and load a ULP program something like:
```
with esp.ulp.ULP() as u:
    u.ADC(u.r1, 0, 0)
    u.ST(u.r1, u.r0, 0)
    u.ADD(u.r0, u.r0, 1)
    u.JUMPR('end', 1024, u.LT)
    u.WAKE()
    u.label('end')
```
(yeah, I haven't looked much this yet, so the above is just an idea of how it might work)

nickzoic, 2018-01-30:
See https://github.com/micropython/micropython/pull/3578 for work on this.
I'd still like to implement an interface like the above.

closed 2018-01-30 by nickzoic

ThomasWaldmann, 2018-03-21:
@nickzoic btw, I implemented an ULP assembler in micropython:

https://github.com/ThomasWaldmann/py-esp32-ulp/

Just can't test it yet on ESP32 without the load/run infrastructure of https://github.com/micropython/micropython/pull/3578 .