micropython/micropython-esp32 #158: Change REPL uart pins

BasedOnTechnology, 2017-08-07:
I see that the machine.uart implementation allows one to change the UART pins, and all UARTs can go through the GPIO matrix. Is there any way for one to change which pins UART0 (REPL) is connected to?

I realize that this is a bit niche, but I could see a few real-world uses (also I need it :wink:).

MrSurly, 2017-08-07:
@BasedOnTechnology 

I don't think µPy exposes the GPIO matrix in Python directly.  Some things (e.g. HW SPI) utilize it.  I suppose the `UART` class initializer could be extended to accept `rx_pin` and `tx_pin`.

BasedOnTechnology, 2017-08-07:
@MrSurly Mainly what I'm looking for is the ability to switch the REPL to UART1, and use UART0 with machine.uart. I tried digging through the code, but I couldn't find where the uart was defined with respect to the REPL.

nickzoic, 2017-08-07:
> @BasedOnTechnology[1]
> thread[3].>

I, uh, think it already does as the `rx`and `tx` parameters to the
constructor ...
... but looking at it, it might not be doing those steps in the right
order ... `machine_uart_init` sets usp `self->tx`(etc) from the args
but `machine_uart_make_new` has already configured the UART to the
default pins.
In any case, I think this is the right way to do it, rather than futzing
with the matrix directly.  We may want to write some code which handles
collisions in pin usage though (?)
-----Nick

MrSurly, 2017-08-08:
> Mainly what I'm looking for is the ability to switch the REPL to UART1

@BasedOnTechnology Looks like UART 0 is already hard-coded dedicated to the repl, as seen in `esp32/uart.c`.

nickzoic, 2017-08-08:
Yeah, I think we could maybe fix that too ... I started off messing around with a telnet server [1] for the REPL  and some kind of generic REPL multiplexor would make sense I think, so that rrepl / webrepl can hook into the REPL mulitiplexor.  Then if your code seizes control of UART 0 it can be disconnected gracefully and the other connections stay alive.

[1] https://github.com/nickzoic/micropython-esp32/tree/esp32-rrepl , don't look too closely at the code, it's rather sketchy but I'm thinking of a tinyssh server instead anyway

dpgeorge, 2017-08-08:
When machine.dupterm() is implemented it will allow you to turn off the REPL from UART0 (and redirect it elsewhere if needed).  See https://github.com/micropython/micropython/pull/3144 for some recent discussion around the dupterm function.

BasedOnTechnology, 2017-08-11:
I don't actually need this anymore, (switched pins in hardware) and since UART already takes pin numbers, I think I'll close this since dupterm would supersede it.

closed 2017-08-11 by BasedOnTechnology

BasedOnTechnology, 2017-09-17:
For citizens of the future: The relevant lines are in sdkconfig.h, modify like so:

`#define CONFIG_CONSOLE_UART_BAUDRATE 115200
#define CONFIG_CONSOLE_UART_CUSTOM 1
#define CONFIG_CONSOLE_UART_CUSTOM_NUM 0
#define CONFIG_CONSOLE_UART_NUM 0
#define CONFIG_CONSOLE_UART_TX_GPIO 17
#define CONFIG_CONSOLE_UART_RX_GPIO 5`