micropython/micropython-esp32 #163: wifi: invalid magic number 3ffcf140

NyxMC, 2017-08-14:
After following setting up the toolchain and the esp-idf, I built the micropython firmware and flashed it on my esp32:
```
make -C mpy-cross
cd esp32
make
make erase
make deploy
```
I can connect to it and besides a `OSError: [Errno 2] ENOENT` everything seems to work fine. But when I import `network`, I get this error:
```
>>> import network
E (174028) wifi: invalid magic number 3ffcf140, call WIFI_INIT_CONFIG_DEFAULT to init config
E (174028) wifi: wifi_init 1361 ret=258
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: Wifi Invalid Argument
```
Recompiling and reflashing the firmware did not fix this issue. Has anyone experienced something similar or am I the only one?

robert-hh, 2017-08-14:
No, I did not see that. After you raised that question in the, I tried again to build & flash a sparkfun esp32 and a Wemos Lolin32 board. No error on import the network module. I'm using the hash value 4ec2abbf23084ac060679e4136fa222a2d0ab0e8

The error `OSError: [Errno 2] ENOENT`is caused by a missing main.py.

dpgeorge, 2017-08-14:
> Recompiling and reflashing the firmware did not fix this issue.

Did you try erasing the flash (make erase) before flashing the firmware?

NyxMC, 2017-08-14:
> Did you try erasing the flash (make erase) before flashing the firmware?
Yes, I did. The strange thing is that the compiled binaries from [MicroPython.org](https://micropython.org) work just fine.

kermit213, 2017-08-14:
+1 i have the same problem, i`am using esp32 module in custom board, but on nodemcu like board with esp32 problem occurs too. More details i postem on micropython forum: https://forum.micropython.org/viewtopic.php?f=18&t=3669

NyxMC, 2017-08-14:
@kermit213 Out of curiosity, on which OS have you compiled the firmware?

kermit213, 2017-08-14:
my is Ubuntu 15.04 ;)

nickzoic, 2017-08-15:
Works fine for me with:

Micropython 3ec00f0d2e518ee9c2fb248751a1e0cd179d60ef
ESP-IDF espressif/esp-idf@4ec2abbf23084ac060679e4136fa222a2d0ab0e8
toolchain from https://dl.espressif.com/dl/xtensa-esp32-elf-linux64-1.22.0-61-gab8375a-5.2.0.tar.gz

Just flashed onto both Rev 0 and Rev 1 silicon and works no problems:
```
MicroPython v1.9.1-397-g3ec00f0d on 2017-08-15; ESP32 module with ESP32
Type "help()" for more information.
>>> import network
I (126633) wifi: wifi firmware version: f092575
I (126633) wifi: config NVS flash: enabled
I (126633) wifi: config nano formating: disabled
I (126633) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (126643) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (126663) wifi: Init dynamic tx buffer num: 32
I (126663) wifi: Init dynamic rx buffer num: 64
I (126663) wifi: wifi driver task: 3ffd79a4, prio:23, stack:4096
I (126673) wifi: Init static rx buffer num: 10
I (126673) wifi: Init dynamic rx buffer num: 0
I (126673) wifi: Init rx ampdu len mblock:7
I (126683) wifi: Init lldesc rx ampdu entry mblock:4
I (126683) wifi: wifi power manager task: 0x3ffde454 prio: 21 stack: 2560
I (126693) wifi: wifi timer task: 3ffdf4bc, prio:22, stack:3584
W (126693) phy_init: failed to load RF calibration data (0x1102), falling back to full calibration
I (126853) phy: phy_version: 355.1, 59464c5, Jun 14 2017, 20:25:06, 0, 2
I (126863) wifi: mode : null
```
Make sure you've done `git submodule update --recursive` in the esp-idf directory *after* checking out the appropriate ESP-IDF version.  This is what I have:
```
$ git show --oneline
4ec2abbf Merge branch 'feature/some_refactor_for_tcpip_adapter' into 'master'

$ git submodule status
 7132505b00d2dd57f48478e75efa636021919aae components/aws_iot/aws-iot-device-sdk-embedded-C (v2.1.1-7-g7132505)
 30a3ab11c441630e3e76fbf30212e546db377e8e components/bt/lib (heads/master-22-g30a3ab1)
 6468887a12666f88b8704d797fc176cd4f40ee4c components/coap/libcoap (v4.1.1-401-g6468887)
 33fcdfb69abc3c21a9cc8ac2422f3747a600efb2 components/esp32/lib (heads/master-118-g33fcdfb)
 ffde3f07b3ed735b6c9b9dfd0c12073a081444c1 components/esptool_py/esptool (v2.0.1)
 70170c28c844a4786e75efc626e1aeebc93caebc components/libsodium/libsodium (1.0.12)
 14222e062d77f45321676e813d9525f32a88e8fa components/micro-ecc/micro-ecc (heads/master)
 2f146e4d4cfe895d65599b87df7d847435f0e1b4 components/nghttp/nghttp2 (v1.22.0)
```

NyxMC, 2017-08-15:
@nickzoic wow. That's it! I followed the instructions of the ESP-IDF before reading the MicroPython-ESP32 docs, so I did forget the `git submodule update --recursive` after checking out. Someone should put that in the docs :smile:.
Thanks for the great support!
@kermit213 Was this also the cause of your problem?

closed 2017-08-15 by NyxMC