micropython/micropython-esp32 #165: Hard time to upload files with ampy and related sw

Mat-Alm, 2017-08-18:
Hi! I had build the last version (with correct idf version) and can get it booting ok. But I can't upload ant files to the device I already tried 2 distinct boards and most of the time I can't even list the files on the devices or read/write most of the times I get errors that it can't open raw repl. With esp8266 I get no problems at all with the same upload softwares

robert-hh, 2017-08-18:
It look like the file system is corrupted. Please try to erase the flash with "make erase" and reload the firmware with "make deploy".
P.S.: Question like this may also be asked at the MicroPython forum https://github.com/micropython/micropython. There is a special section for this ESP32 port, where you also can find hints.

BasedOnTechnology, 2017-08-18:
@Mat-Alm I also had the "can't open repl" problem when I had a terminal program connected to the port simultaneously; closing the serial terminal fixed the issue for me.

nickzoic, 2017-10-04:
Can you connect to the REPL with miniterm.py or similar serial terminal program?

ThomasWaldmann, 2017-11-28:
Not sure if it is it same issue than OP, but also having lots of pain with ampy.

Mostly, it does simply not work. Sometimes hangs for a long time, sometimes aborting with all sorts of exceptions.

What I found out is than it seems to stumble over the stuff the wifi subsystem outputs when connecting.

So, can the wifi log be silenced?

MrSurly, 2017-11-28:
I feel I have to chime in here.

There's something about the WiFi subsystem that screws up the state of the UART RX ring buffers when it starts.  I've had it happen in MicroPython, but it's not a MP problem.  The C project I'm working on now has the same issue, and I've gone with the work-around of not starting Wifi until after I'm done using the UART.

To be clear: I **do not** mean that the WiFi logging is messing up the parsing of console output (though that can happen).  What I mean is there's some sort of deep-seated bug in the IDF (or hardware?) that mucks up the UART when WiFi starts.  Seems to be OK if the UART is basically idle, but if it's receiving data, it can get screwed up, and it remains screwed up until a restart.

ThomasWaldmann, 2017-11-28:
@MrSurly ah, interesting. Now we have >= 2 problems. :-D

I added some debug code to the place where it throws the exception about the repl mode (and i just read some more bytes, not just the 2 that did not match). Then I saw it is the same stuff as seen on the serial console when initing the wifi.

ThomasWaldmann, 2017-11-28:
So, how is everybody working with the esp32 (except the mentioned workaround of not starting the wifi, which is somehow not an option if one wants to develop/debug network apps)?

Is there something more reliable than ampy run / put to run code on the esp32 / upload code to it?

(I am rather new to ESP and micropython, started yesterday with esp32).

MrSurly, 2017-11-28:
I just just https://github.com/nickzoic/mpy-utils, but it is slow (sorry @nickzoic).

nickzoic, 2017-11-28:
I can make it slower ... :-)

Yeah, it is very limited.  I have many pie-in-the-sky ideas for how we
could improve this situation, but they're still fermentingOne big limit is the serial speed, but pushing that any higher tends to
cause problems for users on breadboards.I was thinking of adding a "zmodem" like mode to the repl to simplify
file transfers rather than the current repl-bashing approach.
When mpy-watch works it works rather well, sending only the diffs.  It
is a more complicated beast though, so more prone to getting confused.
I think the key to this is going to be, when in raw repl mode, to hook
the error/log output which normally goes straight to the UART and either
hide it or escape it in such a way that a raw repl client can not be
confused by it.  This is possible in esp idf and I've made some
preliminary progress.

dpgeorge, 2017-11-29:
> So, can the wifi log be silenced?

I just merged #112 which added esp.osdebug().  You can now silence the logging using esp.osdebug(None).

nickzoic, 2017-11-29:
OK, I'm going to add that into mpy-utils as the first thing it runs :-)
What did you think about the idea of adding that to raw mode in general?
Too weird?

dpgeorge, 2017-11-29:
> What did you think about the idea of adding that to raw mode in general?

It would make sense... but even getting into raw mode may not work if the debug messages are there.

PikWay, 2017-11-29:
One of my painfull issues with ampy is solved with patch & setting osdebug to None. Thanks! 
It was with nested dirs - I was unable to rmdir from ampy (it was ending with error) when ESP was connected with STA_IF, os.listdir was showing that dir is gone, but when I was trying to put it again using ampy it was failing with EXISTS.

bboser, 2017-12-05:
Seeing wifi in this thread - is there a way to get a REPL over wifi on the ESP32 with the official micro python firmware? 
(Or is it merely that wifi and uart interact in a way that may mess up REPL over serial).

BTW: REPL over wifi (telnet) works reliably with rshell and the loboris firmware. I have not noticed interaction between uart and wifi (though there are other bugs, and incompatibilities with "standard" micro python).

nickzoic, 2017-12-05:
I have a telnet implementation in a branch: it's very minimal but works
well, I don't know if that's something Damien would consider merging.

bboser, 2017-12-05:
I'd love to try it!
Where can I find it? And are there any instructions to use it?

Many thanks!!!

nickzoic, 2017-12-05:
https://github.com/nickzoic/micropython-esp32/tree/esp32-rrepl
but it's quite out of date now, I'll rebase it onto current master, uh,
probably tomorrow.

bboser, 2017-12-06:
Great. I’ll wait for the rebase.

Just out of curiosity - rrepl.c uses freertos & presumably creates a task
for telnet.

Many thanks!
Bernhard

Avi-TelnT, 2017-12-06:
@nickzoic 
Thank you. 
Can you also offer how to embeded this module into https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo
?

bboser, 2017-12-06:
I don’t think that’s needed as telnet is already part of loboris
micropython.

Bernhard

dpgeorge, 2019-09-25:
It should be possible to use the "-f" flag to the latest pyboard.py script to access the filesystem on the esp32 over the serial/UART.

closed 2019-09-25 by dpgeorge