micropython/micropython-esp32 #168: README.md Please provide a firmware installation "howto".

peterhinch, 2017-08-28:
Firmware installation has caused problems to forum users, in particular the 0x1000 offset which seems to be undocumented. A one-liner in README.md recommending an esptool invocation would save a lot of confusion :)

MrSurly, 2017-08-28:
If you're building it from this repo, the  README states to use `make deploy`, which does work.  Running this with `V=1` set will show the actual esptool command being used.

The [download](http://micropython.org/download#esp32) page for the ESP32 shows the 0x1000 offset on the command for flashing.

peterhinch, 2017-08-29:
Fair enough. It arose with users installing the standard build and as you say the download page is clear enough. I'll point it out if it arises again in the forum.

closed 2017-08-29 by peterhinch