micropython/micropython-esp32 #170: error while make for esp32

aronipani, 2017-08-30:
I am getting the following error while trying to build micropython for esp-wroom-32
command: 
$ cd esp32
$ make

error: this is in mac

CC /Volumes/myfolder/esp/esp-idf/components/wpa_supplicant/src/crypto/bignum.c
CC /Volumes/myfolder/esp/esp-idf/components/wpa_supplicant/src/crypto/crypto_internal-modexp.c
CC /Volumes/myfolder/esp/esp-idf/components/wpa_supplicant/src/crypto/crypto_internal-cipher.c
CC /Volumes/myfolder/esp/esp-idf/components/wpa_supplicant/port/os_xtensa.c
LINK build/application.elf
build/modnetwork.o:(.literal.esp_initialize+0xc): undefined reference to `esp_wifi_init'
build/modnetwork.o: In function `esp_initialize':
modnetwork.c:(.text.esp_initialize+0x2b): undefined reference to `esp_wifi_init'
make: *** [build/application.elf] Error 1

Can you please advise. 
thanks for your help.

robert-hh, 2017-08-30:
Try to checkout the version of the IDF expected by the micropython firmware. The hash tag of the expected version is told when the make script starts. Just copy that string, go into the esp-idf directory and run the commands:

`git checkout <copied hash tag>`
`git submodule update --init`

aronipani, 2017-08-30:
Thanks, I had to run the git submodule update --init , again and now it worked. Thanks for your help.

closed 2017-08-30 by aronipani