micropython/micropython-esp32 #171: Explaining wifi debug messages

binloan, 2017-08-30:
I built the esp32 port on my HUZZAH32 and when I successfully connected to wifi I get constantly the following warning:

`W (544828) wifi: noTIM!!`

What exactly does this mean? Can someone explain this to me? I guess I can ignore it, but I still would like to know its origin. Thanks

MrSurly, 2017-08-30:
This error message comes from the binary portion of the ESP IDF, which MicroPython relies.  You'd probably be better off asking on the IDF forums.

nickzoic, 2017-09-12:
My guess is that it might be something to do with this: https://en.wikipedia.org/wiki/Traffic_indication_map but since it's part of the binary firmware blobs there's no easy way to tell.

@binloan Probably your best bet is to write a small snippet of C which connects to your WiFi and see if that reproduces the problem: this will let you report it to the ESP-IDF issue tracker at https://github.com/espressif/esp-idf/issues

I'm going to close this one since there's nothing the MicroPython project can do about it ...

closed 2017-09-12 by nickzoic