micropython/micropython-esp32 #173: ADC reference voltage - HUZZAH32

binloan, 2017-08-31:
Hello,

while experimenting with the HUZZAH32 (ESP-WROOM-32) I found a strange behaviour regarding the ADCs. Adafruit uses the I35 as a battery voltage pin. They added a 100k voltage divider which allows measuring and monitoring the current voltage of the battery.
However I found that this ADC returns always 4095. It looks like the internal reference voltage is set to 1.24V rather than 3.3V. (Because at that point it starts to decrease)

Is this a bug, or did I miss setting another value while defining the ADC pin? Can someone confirm this behaviour? Thanks.

robert-hh, 2017-08-31:
The internal ADC of the ESP32 range is specified as 0..1V, practically it is 0.05 .. 1.1 V, and may vary.
This may be considered as a bug, but not of MicroPython. Maybe it is better to ask these questions at the pycom user forum, where more people may respond. https://forum.pycom.io/, or at the esp user forum https://www.esp32.com/

chrismas9, 2017-08-31:
You need to set the attenuation to 11db. This will divide the input by approx 3.6 and allows 0 to 3.3v input range with some non linearity. See:

https://esp-idf.readthedocs.io/en/v2.0/api/peripherals/adc.html

https://www.esp32.com/viewtopic.php?t=1045

binloan, 2017-09-01:
@chrismas9 That's a good point. I didn't notice that this option was even there.

This solved the issue. Just for the record. On my ADC object I can set the attenuation via the atten() function. It was too obvious to be seen. Thank you both.

closed 2017-09-01 by binloan