micropython/micropython-esp32 #174: UART.writechar() implementation

BasedOnTechnology, 2017-09-01:
Since there's no UART.writechar() implementation, it is currently impossible to use protocols like Xmodem.

--
I'm pretty sure this is expected behavior,

If I do this:

`sam_uart.write('\x7f')`

or this:

`sam_uart.write(chr(127))`

And then check the output on a hex viewer, I get 

`0x7f`

Which is correct. If I do this:

`sam_uart.write(chr(128))`

or this:

`sam_uart.write('\x80')`

I get 

> 0xc2 0x80

If I do this:

`sam_uart.write(chr(255))`

I get 

> 0xC3 0xBF

However, on an Arduino, this isn't the behavior - Serial.write, when given one byte, writes that byte verbatim.

Had me tearing my hair out.

MrSurly, 2017-09-01:
It's sending the UTF8 encoding. What happens when you use a 'bytearray'?

write(b'\x80')

<https://github.com/micropython/micropython-esp32/issues/174>, or mute the
thread

BasedOnTechnology, 2017-09-01:
@MrSurly That works! However, bytearray doesn't work - it has the same problem.

MrSurly, 2017-09-01:
I'm confused, did it work, or didn't it?

BasedOnTechnology, 2017-09-01:
@MrSurly Sorry, b'\80' works, but bytearray() doesn't.

BasedOnTechnology, 2017-09-01:
@MrSurly For the record, bytes([0x80]) worked. Thanks for your help.

closed 2017-09-01 by BasedOnTechnology

dpgeorge, 2017-09-06:
> but bytearray('\x80') doesn't.

That's because that piece of code converts the string "\x80" to a utf8 encoded byte stream.  You can use `bytearray([0x80])` or `bytearray(b'\x80')`, but the most efficient thing is `b'\x80'`.

BasedOnTechnology, 2017-09-06:
@dpgeorge Yup, serves me right to not learn the workings of python 3 strings before raising an issue :)

nickzoic, 2017-09-06:
I'd argue this is a bug in MicroPython, because bytearray('\x80') should
throw a TypeError in Python3.
```
Python 3.5.2 (default, Nov 17 2016, 17:05:23)

>>> bytearray([0x80])
bytearray(b'\x80')

>>> bytearray(b'\x80')
bytearray(b'\x80')

>>> bytearray('\x80')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: string argument without an encoding

>>> bytearray('\x80', 'utf-8')
bytearray(b'\xc2\x80')
```

... throwing these exceptions is annoying, but less annoying than not
throwing them!

dpgeorge, 2017-09-06:
> I'd argue this is a bug in MicroPython, because bytearray('\x80') should throw a TypeError in Python3.

Yes good point.