micropython/micropython-esp32 #175: usocket.getaddrinfo() returns an empty list

peterhinch, 2017-09-05:
Using today's build:
```python
MicroPython v1.9.1-477-g75ead22c on 2017-09-05; ESP32 module with ESP32
Type "help()" for more information.
>>> 
>>> import usocket as socket
I (175278) modsocket: Initializing
>>> socket.getaddrinfo('iot.eclipse.org', 1883)
[]
>>> socket.getaddrinfo('google.com', 80)
[]
>>> 
```

peterhinch, 2017-09-05:
Sorry - I'm too used to the ESP8266. It works fine.

closed 2017-09-05 by peterhinch

ulidtko, 2023-08-07:
> Or does `getaddrinfo` always return something?

On success — **yes**. Per both [POSIX][] and [IETF RFC 2553][]:

> ... and, in absence of errors, **one or more results** shall be returned.

> Upon successful return a pointer to a linked list of **one or more** addrinfo structures is returned ...

As an implementor of `getaddrinfo` — you would signal "network down" / "DNS unavailable" conditions by *error return*: e.g. `EAI_SERVICE`, `EAI_AGAIN`, `EAI_SYSTEM` (and in python: exceptions) — not by "successfully" returning empty list. 

[POSIX]: https://pubs.opengroup.org/onlinepubs/9699919799/functions/getaddrinfo.html
[IETF RFC 2553]: https://www.rfc-editor.org/rfc/rfc2553

dpgeorge, 2023-08-16:
The current version of the esp32 port of MicroPython is found at https://github.com/micropython/micropython/, and it does now raise an exception (an OSError) if there is no network connection when calling `getaddrinfo`.

ulidtko, 2023-08-16:
Hey @dpgeorge, thanks for your comment... Glad to hear.  I'd recommend [archiving][] this repo if it's by now of historical interest only; it's a bit too easy to miss the readme notice, because issues also get indexed by search engines. 

[archiving]: https://docs.github.com/en/repositories/archiving-a-github-repository/archiving-repositories

dpgeorge, 2023-09-06:
I will archive this repo.