micropython/micropython-esp32 #176: Mesh Network

pedrofelipest, 2017-09-06:
How to create mesh network? Has it been implemented?

MrSurly, 2017-09-06:
For MicroPython ESP32? No.   Probably not anytime soon.

nickzoic, 2017-09-12:
I went and created this issue for ESP-IDF to add 802.11 Ad-Hoc Networks, but I wouldn't hold your breath.
https://github.com/espressif/esp-idf/issues/989

nickzoic, 2017-09-29:
@pedrofelipest ESP-NOW protocol might also be of interest to you.
I'm currently working on getting that implemented for MicroPython.

pedrofelipest, 2017-09-29:
Thanks @nickzoic, I will check this implementation.

nickzoic, 2018-01-03:
(FYI: Espressif aren't likely to support WiFi AdHoc but have other things on the go, including WiFi Direct and ESP-Mesh.  I've closed the issue above.)

pedrofelipest, 2018-01-16:
@nickovs the ESP-NOW do you have a maximum amount of devices on the network that is mounted?

closed 2019-01-16 by dpgeorge

60999, 2020-11-29:
Is there any update progress?