micropython/micropython-esp32 #178: SHA1 and uhashlib

rdagger, 2017-09-11:
I tried to import uhashlib and it doesn't appear to be implemented.
I'm using the latest daily build: esp32-20170911-v1.9.2-270-g14fb53e0.bin
Is there currently any support for sha1?

MrSurly, 2017-09-11:
```python
MicroPython v1.9.2-270-g14fb53e0 on 2017-09-11; ESP32 module with ESP32
Type "help()" for more information.
>>> import binascii, hashlib
>>> hash = hashlib.sha1('mango')
>>> binascii.hexlify(hash.digest()) 
b'934aae49f648ed870c9c421829f4cece6643cf86'
>>> 
```

Works for me, and the hash is correct. Can you send any code that isn't working?

MrSurly, 2017-09-11:
If this is resolved, please close.

rdagger, 2017-09-11:
The issue I had was the [docs](https://docs.micropython.org/en/latest/pyboard/library/uhashlib.html) refer to the library as uhashlib instead of just hashlib.  Thanks again.

MrSurly, 2017-09-11:
I noticed that as well.  All of the libraries that start with 'u' should
import without the 'u'.

I think that this might be a legitimate issue with aliasing. I'll look into
it.

nickzoic, 2017-09-12:
Yeah, just rebuilt and `uhashlib` won't import
```
MicroPython v1.9.2-270-g14fb53e0 on 2017-09-12; ESP32 module with ESP32
Type "help()" for more information.
>>> import uhashlib
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: no module named 'uhashlib'
>>> import hashlib
>>>
```
It also doesn't appear in `help('modules')`

OK, so the new esp32-specific module was in MICROPY_PORT_BUILTIN_MODULE_WEAK_LINKS but wasn't in MICROPY_PORT_BUILTIN_MODULES, is all.

MrSurly, 2017-09-30:
@nickzoic  Is this still an issue?

nickzoic, 2017-10-01:
Ok, merged #179 and closing this ...

closed 2017-10-01 by nickzoic