micropython/micropython-esp32 #180: Cannot build - stop at LINK build/bootloader.elf

ubidefeo, 2017-09-12:
I have followed all the instructions.
Using a vagrant provisioned machine
Distributor ID:	Ubuntu
Description:	Ubuntu 14.04.5 LTS
Release:	14.04
Codename:	trusty

it all seems to work fine, but when it comes to building the bootloader the ESPIDF variable gets expanded into a path and it messes it up

```
LINK build/bootloader.elf
xtensa-esp32-elf-gcc: error: build/bootloader//home/vagrant/esp-idf//components/bootloader/src/main/bootloader_start.o: No such file or directory
```

at this point probably the ESPIDF should not be used in the path, and I think it comes from ESPCOMP in BOOTLOADER_OBJ

Can anyone help out?
I'd really like to build this, as I'm having issues with the precompiled binary's file system `/flash` mount.

Thanks :)

MrSurly, 2017-09-12:
> it all seems to work fine, but when it comes to building the bootloader the ESPIDF variable gets expanded into a path and it messes it up

That's normal. 

Few of things to check: 

1. Are you using the correct commit hash for the IDF?  If you get a hash warning when you build, that could be a problem.
2. After ensuring #1, make sure you do `git submodule update --init --recursive`
3. Also, make sure you've done `git submodule update --init --recursive` from within the MicroPython repo.

ubidefeo, 2017-09-12:
hi @MrSurly 
I have indeed done a recursive submodule update.

I guess I'll scrap this box and start from scratch tomorrow :D 
thanks for the tips

u.

MrSurly, 2017-09-12:
Write down your steps, post here, and I'll try it.

nickzoic, 2017-09-12:
Ummm, yeah, the ESPIDF variable being inserted into the build path is
what causes #142so I'm pretty keen to clean it up, but it's normal and works fine on
*nix.
eg: I have a file
`/home/nick/Work/esp32/micropython-
esp32/esp32/build/bootloader/home/nick/Work/esp32/esp-
idf/components/bootloader/src/main/bootloader_start.o`
on my computer.  Does it look like your build system is creating these
directories?

MrSurly, 2017-09-13:
> I'm pretty keen to clean it up, but it's normal and works fine on *nix.

Right, and @ubidefeo states he's using Ubuntu, which is why I'm puzzled it doesn't work for him.

I'm OK with trying this out in vagrant myself, and troubleshooting.

Also, it's odd that it fails at the link stage, especially since the `Makefile` explicitly mentions the source file, so it's not as if it's just a missing .c file that didn't get swept up in a glob.

MrSurly, 2017-09-13:
@ubidefeo Does your build line look something like this?

`ESPIDF=/home/vagrant/esp-idf make`

ubidefeo, 2017-09-13:
@nickzoic , the directories get created, but nothing's in them after the error is thrown
```
vagrant@vagrant-ubuntu-trusty-64:~/micropython-esp32/ports/esp32/build/bootloader/home/vagrant/esp-idf/components/bootloader/src/main$ ls -la
total 8
drwxrwxr-x 2 vagrant vagrant 4096 Sep 13 05:36 .
drwxrwxr-x 3 vagrant vagrant 4096 Sep 13 05:36 ..
```

@MrSurly 
> Does your build line look something like this?
> ESPIDF=/home/vagrant/esp-idf make

I use a `makefile` in which I setup a single variable

```
ESPIDF = $(HOME)/esp-idf/
include Makefile
```
`$(HOME)/esp-idf/` expands to `/home/vagrant/esp-idf/` and I believe it's correct.
What's strange to me is that it builds correctly up until the bootloader linking phase.

```
CC /home/vagrant/esp-idf//components/spi_flash/spi_flash_rom_patch.c
CC /home/vagrant/esp-idf//components/soc/esp32/rtc_clk.c
CC /home/vagrant/esp-idf//components/soc/esp32/rtc_time.c
CC /home/vagrant/esp-idf//components/micro-ecc/micro-ecc/uECC.c
LINK build/bootloader.elf
xtensa-esp32-elf-gcc: error: build/bootloader//home/vagrant/esp-idf//components/bootloader/src/main/bootloader_start.o: No such file or directory
make: *** [build/bootloader.elf] Error 1
```

I'm having my morning coffee and then I'll give it a go on a fresh machine :)

ubidefeo, 2017-09-13:
so...
after doing a bit of this
```
checkout 4ec2abbf23084ac060679e4136fa222a2d0ab0e8
git submodule update --recursive
```

I get past that error, but then another can of worms opens
```
LINK build/application.elf
/home/vagrant/esp-idf//components/esp32/lib/libpp.a(wdev.o): In function `wDev_ProcessTxop':
(.iram1+0x128): undefined reference to `os_timer_disarm'
/home/vagrant/esp-idf//components/esp32/lib/libpp.a(wdev.o): In function `wDev_ProcessTxop':
(.iram1+0x12c): undefined reference to `os_timer_arm'
/home/vagrant/esp-idf//components/esp32/lib/libpp.a(wdev.o): In function `wDev_ProcessTxop':
(.iram1+0x161): undefined reference to `os_timer_disarm'
/home/vagrant/esp-idf//components/esp32/lib/libpp.a(wdev.o): In function `wDev_ProcessFiq':
(.iram1+0x18d): undefined reference to `os_timer_arm'
build/modnetwork.o:(.literal.esp_initialize+0xc): undefined reference to `esp_wifi_init'
build/modnetwork.o: In function `esp_initialize':
```

any of this happened to anyone?

ubidefeo, 2017-09-13:
scratch that!
```
$ make clean
$ git submodule init lib/berkeley-db-1.xx
$ git submodule update
```

I got a build!
```
LINK build/application.elf
   text	   data	    bss	    dec	    hex	filename
 705611	 197080	 138440	1041131	  fe2eb	build/application.elf
Create build/application.bin
esptool.py v2.0.1
Create build/firmware.bin
bootloader     13216
partitions      3072
application  1024976
total        1090512
```

projectgus, 2017-09-13:
I'm coming to this thread a bit late, but to help explain: in a recent IDF update the `esp-idf/components/bootloader/src/` directory was renamed to `esp-idf/components/bootloader/subproject`. So if you take a recent IDF master branch it will have a different directory name.

nickzoic, 2017-09-13:
No worries (and thanks @projectgus), we keep our ESP-IDF version pinned for a reason :-)
Closing this one ...

closed 2017-09-13 by nickzoic