micropython/micropython-esp32 #182: Error in network.connect

pedrofelipest, 2017-09-19:
after the esp32 connects to a network it hangs.

`def do_connect():
    import network
    sta_if = network.WLAN(network.STA_IF)
    if not sta_if.isconnected():
        print('connecting to network...')
        sta_if.active(True)
        sta_if.connect('ap1', pswd1234')
        while not sta_if.isconnected():
            pass
    print('network config:', sta_if.ifconfig())
do_connect()`

I (15359) wifi: pm start type:0

MrSurly, 2017-09-19:
Does it "hang" in that CTRL-C doesn't work, or is it simply in the `while` loop waiting for a connection?  It's likely the latter -- could you instrument that code to check?

The snippet attached works fine for me.

robert-hh, 2017-09-19:
What I use is a variant of the sample script:
```
def do_connect():
    import network
    import time

    wlan = network.WLAN(network.STA_IF) # create station interface
    wlan.active(True)       # activate the interface
    if not wlan.isconnected():      # check if the station is connected to an AP
        wlan.connect('ssid_AP', 'pwd_AP') # connect to the AP (Router)
        for _ in range(10):
            if wlan.isconnected():      # check if the station is connected to an AP
                wlan.ifconfig(("my_ip", "255.255.255.0", "gateway", "DNS"))
                break
            print('.', end='')
            time.sleep(1)
        else:
            print("Connect attempt timed out\n")
            return
    print('\nnetwork config:', wlan.ifconfig())

do_connect()
```
But you snippet should work too. Are you sure that you used the proper credentials?

pedrofelipest, 2017-09-19:
Yes, I used the credentials.

I implemented this code.

`
import sys
import network
import gc
import time
import socket

def do_connect():
    import network
    import time

    wlan = network.WLAN(network.STA_IF) # create station interface
    wlan.active(True)       # activate the interface
    if not wlan.isconnected():      # check if the station is connected to an AP
        wlan.connect('IFCE-LEM', 'lem-wifi') # connect to the AP (Router)
        for i in range(10):
            if wlan.isconnected():      # check if the station is connected to an AP
                break
            print('.', end='')
            time.sleep(1)
        else:
            print("Connect attempt timed out\n")
            return
    print('\nnetwork config:', wlan.ifconfig())
`

The following appears:

>>> do_connect()
I (12266) wifi: mode : sta (30:ae:a4:05:da:98)
I (12266) wifi: STA_START
..I (13596) wifi: n:11 2, o:1 0, ap:255 255, sta:11 2, prof:1
I (14156) wifi: state: init -> auth (b0)
I (14156) wifi: state: auth -> assoc (0)
I (14156) wifi: state: assoc -> run (10)
I (14196) wifi: connected with IFCE-LEM, channel 11
I (14196) wifi: event 4
..I (15406) event: ip: 10.103.4.32, mask: 255.255.248.0, gw: 10.103.0.1
I (15406) wifi: GOT_IP

network config: ('10.103.4.32', '255.255.248.0', '10.103.0.1', '189.90.16.20')
>>> I (24156) wifi: pm start, type:0

Then it gets blocked for anything.

robert-hh, 2017-09-19:
That means that you connected to your AP.  What else follows in the code? Is that snippet in a small file which you import ?

B.t.w.: Code blocks in this board must be enclosed in a pair of three backticks.

pedrofelipest, 2017-09-19:
após isso trava. Não tenho mais acesso ao terminal, nem mesmo CTRL+C

robert-hh, 2017-09-19:
How do you connect to the board?

pedrofelipest, 2017-09-19:
By USB. I use the terminal in Ubuntu with the picocom

robert-hh, 2017-09-19:
That's strange. Do you run any other code that this small script? 
Just for interest: which esp board do you use?

pedrofelipest, 2017-09-19:
Yes, I am using Bluetooth normally in other code normally.
It's an ESP32.

robert-hh, 2017-09-19:
So at the moment you execute this script, you already run other code, or do you try this littel script by itself after booting up? I'm asking because running this script causes more memory to be allocated, and that may be a problem if the memory is almost used up. And, for completeness, which version of the firmware are you using. This is told by:
import uos
uos.uname()

pedrofelipest, 2017-09-19:
sysname='esp32', nodename='esp32', release='1.8.6', version='v1.8.6-1699-g6add00f on 2017-09-04', machine='ESP32 module with ESP32'

pedrofelipest, 2017-09-19:
I not have anything in boot.py
I executed this:
`
>>> import network
>>> sta = network.WLAN(network.STA_IF)
>>> sta.active(1)
I (266031) wifi: mode : sta (30:ae:a4:05:df:50)
I (266031) wifi: STA_START
True
>>> sta.connect('IFCE-LEM','lem-wifi')
>>> I (282171) wifi: n:11 2, o:1 0, ap:255 255, sta:11 2, prof:1
I (282721) wifi: state: init -> auth (b0)
I (282731) wifi: state: auth -> assoc (0)
I (282731) wifi: state: assoc -> run (10)
I (282761) wifi: connected with IFCE-LEM, channel 11
I (282761) wifi: event 4
I (284261) event: ip: 10.103.1.254, mask: 255.255.248.0, gw: 10.103.0.1
I (284261) wifi: GOT_IP
I (292731) wifi: pm start, type:0
`

robert-hh, 2017-09-19:
That's strange, since the device connects and gets an IP address. Just as a last resort, try to update to the modes recent version from the MP repository here: http://micropython.org/download/#esp32

pedrofelipest, 2017-09-19:
Este repositório tem a implementação do Bluetooth como no https://github.com/MrSurly/micropython-esp32/tree/dev-bluetooth/esp32?

robert-hh, 2017-09-19:
This does not implement Bluetooth, but that's just a test to sort out other problems. If it works with the "official" build, then there is a problem with MrSurly's build, even if he could run your script.

MrSurly, 2017-09-19:
@robert-hh I actually used an up-to-date `esp32` branch, but let me try it on `dev-bluetooth`.

MrSurly, 2017-09-19:
> then there is a problem with MrSurly's build, even if he could run your script.

@robert-hh Confirmed -- this freezes for me after it connects when using my BLE branch, usually right after seeing `I (76378) wifi: pm start, type:0`.

nickzoic, 2017-09-26:
Oh goodie, time to rebase and bisect :-/.

MrSurly, 2017-09-26:
> Oh goodie, time to rebase and bisect :-/

Yeah, and I'm swamped with other stuff right now.

MrSurly, 2017-09-30:
@pedrofelipest  I recommend closing this issue, since it's not an issue with the mainline ESP32 port, but with my currently un-merged PR.  and there's already an [issue](https://github.com/MrSurly/micropython-esp32/issues/8) for that.

pedrofelipest, 2017-10-02:
Ok @MrSurly. Thanks.

closed 2017-10-02 by pedrofelipest