micropython/micropython-esp32 #184: Minor:  The ESPIDF_CURHASH doesn't work if a submodule is out of date

stestagg, 2017-09-23:
This is kinda minor, but posting for visibility:

I just had the build fail with this message:

```
.../components/esp32/lib/libnet80211.a(wl_cnx.o):(.text.cnx_start_handoff_cb+0x0): undefined reference to `wpa2_machine_start'
```

This is due to some issue with the esp-idf  esp32 lib submodule on master.

As per the check in the makefile:
```
ESPIDF_SUPHASH := 9a26296a0e88a4c3ae27e9c848be970946fff87e
ESPIDF_CURHASH := $(shell git -C $(ESPIDF) show -s --pretty=format:'%H')
ifneq ($(ESPIDF_CURHASH),$(ESPIDF_SUPHASH))...
```

I checked out 9a26296a, and re-ran.  This made the warning message go away, but the real problem was in a submodule, so without running `git submodule update`, it was still using the bad code, but the HASH check on the esp-idf repo didn't catch this.

The link completed once I'd updated the submodules

MrSurly, 2017-09-23:
The way git is implemented, the repo hash doesn't include submodule checkout state (since they're their own repos).  A bit odd that `git checkout` doesn't automatically run `git submodule update`.

It would be kinda neat to have the ESP32 makefile check the hash, and prompt to update the IDF (including submodules), the perform the action if desired.

projectgus, 2017-09-25:
FWIW there's a check we do in the IDF build system to warn on out of date submodules. You probably can't run the exact same check from the Micropython build system, because it relies on each component reporting its associated submodules (and Micropython only has a list of source files).

However the Micropython build could parse the output of `git submodule status --recursive` and look for any line that doesn't start with ` `, then warn on that. It's not a git plumbing command but I can't expect it's likely to change too much.

> A bit odd that git checkout doesn't automatically run git submodule update.

FWIW,  either doing `git checkout --recurse-submodules` or setting the `submodule.recurse` option in git config will select this behaviour.

closed 2019-01-16 by dpgeorge