micropython/micropython-esp32 #193: Timeout on ntptime.time() on esp32

manningt, 2017-10-04:
ntptime.time() works fine on ESP8266 and OSX platforms.

I can't get it to work on an ESP32, as per the following trace:
```
>>> wlan.ifconfig()
('10.0.1.53', '255.255.255.0', '10.0.1.1', '10.0.1.1')
>>> import usocket as socket
␛[0;32mI (75948) modsocket: Initializing␛[0m
>>> NTP_QUERY = bytearray(48)
>>> NTP_QUERY[0] = 0x1b
>>> addr = socket.getaddrinfo("pool.ntp.org", 123)[0][-1]
>>> s = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)
>>> s.settimeout(1)
>>> s.sendto(NTP_QUERY, addr)
48
>>> msg = s.recv(48)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 110] ETIMEDOUT
```
The ESP32 module I'm using can communicate to other servers using urequests, so I don't think its a networking problem.

I do not have a sniffer to determine if the socket._sendto_ is sending the NTP_QUERY correctly.
Adding socket.SO_REUSEADDR did not change things.

Does anyone else have ntptime.time working on an ESP32?  Any suggestions on debugging?

MrSurly, 2017-10-04:
I omitted the `settimeout(1)` call. Looks like it worked, but I didn't parse the return packet.

```python
>>> NTP_QUERY = bytearray(48)
>>> NTP_QUERY[0] = 0x1b
>>> NTP_QUERY
bytearray(b'\x1b\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')
>>> addr = socket.getaddrinfo("pool.ntp.org", 123)[0][-1]
>>> s = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)
>>> s.sendto(NTP_QUERY, addr)
48
>>> msg = s.recv(48)
>>> msg
b'\x1c\x02\x03\xec\x00\x00\x00"\x00\x00\x0f\xb0\xc6\xa9\xd0\x90\xdd\x7f\xb9=\xdf<"Z\x00\x00\x00\x00\x00\x00\x00\x00\xdd\x7f\xbf\xafs\xb3\x04m\xdd\x7f\xbf\xafs\xb8\\\x1c'
```

manningt, 2017-10-05:
Thanks for testing.
I did additional testing and it appears the issue is the NTP_QUERY packet doesn't get sent if it's the first thing performed after STA_IF activation.  RECVing the MSG worked if either (a) a urequest was performed after wlan activation or (b) the NTP_QUERY packet is sent twice.

The timeout appears to work fine.

A secondary issue is the s.close() gets a **OSError: [Errno 22] EINVAL** exception on the ESP32.  I had a loop that would try ntptime.time() multiple times until success.  The first attempt would get a timeout, presumably because the first packet wasn't set.  The remaining attempts after that EINVAL exception because of the s.close() issue.

So instead of using ntptime.time() I'm using the attached function. [get_ntp_time.txt](https://github.com/micropython/micropython-esp32/files/1359124/get_ntp_time.txt)

manningt, 2017-10-06:
Yes, I'm sure that it does not time-out: without the timeout, it just hangs.  I retested and here are the results:
- without the timeout, the `s.recv` never returns
- With `s.settimeout(2)` a timeout does occur on the `s.recv`
- WIth 2 `s.sendto` and no timeout, the `s.recv` gets the msg.

I don't have my esp8266 dev board right now, but I think the same thing (the 1st packet doesn't get sent after WiFi init) might occur on the 8266 port.  I recall the first `ntptime.time()` always got a timeout, hence I put it in a loop to retry.  Its just that on the 8266, the s.close didn't get an exception, whereas the s.close causes an exception on the ESP32.

dpgeorge, 2017-10-06:
It looks like there are some issues here to sort out.  s.close() generally shouldn't raise an exception.

manningt, 2017-10-07:
To reproduce the _OSError: [Errno 22]_ then do the following after `wlan.connect()`:
```
import usocket as socket
NTP_QUERY = '\x1b' + 47 * '\0'
addr = socket.getaddrinfo("pool.ntp.org", 123)[0][-1]
s = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)
s.sendto(NTP_QUERY, addr)
s.sendto(NTP_QUERY, addr)
msg = s.recv(48)
s.close()
```

dpgeorge, 2017-10-10:
I think the timeout issue is just because UDP is an unreliable connection.

The close() issue seems to be a problem with the IDF, reported here: https://github.com/espressif/esp-idf/issues/1094

BasedOnTechnology, 2017-10-25:
@dpgeorge @MrSurly 

I'm getting the following:
```

Socket Open
mbedtls_ssl_handshake error: -7280
Socket has timed out: [Errno 5] EIO
Traceback (most recent call last):
  File "main.py", line 548, in <module>
  File "main.py", line 525, in process_data
  File "main.py", line 145, in send_to_server
OSError: [Errno 22] EINVAL

<another loop>

Sending to server /update
Socket Open

assertion "0" failed: file "../../py/../extmod/modussl_mbedtls.c", line 151, function: socket_new
abort() was called at PC 0x40131a87 on core 0

Backtrace: 0x400873fb:0x3ffce020 0x40087427:0x3ffce040 0x40131a87:0x3ffce060 0x400ecf2b:0x3ffce090 0x400df716:0x3ffce0e0 0x400dbbc1:0x3ffce110 0x400dbc29:0x3ffce130 0x400e8c09:0x3ffce150 0x400df7a4:0x3ffce1f0 0x400dbbc1:0x3ffce220 0x400e8b7f:0x3ffce240 0x400df7a4:0x3ffce2e0 0x400dbbc1:0x3ffce360 0x400e8b7f:0x3ffce380 0x400df7a4:0x3ffce420 0x400dbbc1:0x3ffce450 0x400dbbee:0x3ffce470 0x400f7352:0x3ffce490 0x400f75b5:0x3ffce530 0x400eeea7:0x3ffce550

Rebooting...
���x����xx�������

```

As you can see, if the ssl connection times out, s.close() returns an exception, and then on the next connection attempt, the ESP32 hard-faults (unrecoverable). This is really, really serious for us.

The related code is here:

```
def verify_server(socket):
    certHash = binascii.hexlify(
        hashlib.sha256(socket.getpeercert(True)).digest())
    print("Certificate Hash: " + str(certHash))
    for server in server_dict:
        if certHash in server['fingerprints']:
            return True
    return False

def send_to_server(data, url, server_index=0):
    print("Sending " + str(data) + "to server " + url)
    s = socket.socket()
    print("Socket Open")
    try:
        s.settimeout(30)
        ai = socket.getaddrinfo(server_dict[server_index]['address'], 443)
        addr = ai[0][-1]  
        s.connect(addr)
        s = ssl.wrap_socket(s, cert=sump_cert_2.CERT, key=sump_cert_2.KEY)
        if not verify_server(s):
            print("Certificate doesn't match.")
            s.close()
            return None
        server = server_dict[server_index]
        http_headers = '\r\n'.join([
            "POST {} HTTP/1.1".format(url),
            "Connection: Keep-Alive",
            "Accept: */*",
            "Host: {}".format(server['address']),
            "User-Agent: test",
            "Content-Type: application/x-msgpack",
            "Content-Length: {}".format(len(data)),
            "\r\n"])
        s.write(http_headers)
        s.write(data)
        response = s.read(2048)
    except OSError as e:
        print("Socket has timed out: {}".format(e))
    except Exception as e:
        print("Server connection failed: {}".format(e))
        store_in_backlog(data)
    else:
        return response
        # process_backlog()
    finally:
        s.close()
        print("Socket Close")
    return None
```

Is there any possible workaround for this? 

Thanks!

dpgeorge, 2017-10-26:
In 02f94f7ed55ab69af9ae0fe9b9beeda418277fef I updated the ESP IDF to the latest version and the socket.close() issue for UDP sockets is resolved.  In dba3796e3d2ea6e2d05d16c07170314a6330d73a I symlinked the ntptime.py module so that it's included with the build (following esp8266) and verified that it works correctly.

closed 2017-10-26 by dpgeorge