micropython/micropython-esp32 #197: ESP-Now Protocol

nickzoic, 2017-10-08:
ESP-Now offers a low-level adhoc network over 802.11 frames ... make this available from a nice MicroPython API.

Refs:
http://espressif.com/sites/default/files/documentation/esp-now_user_guide_en.pdf
http://esp-idf.readthedocs.io/en/latest/api-reference/wifi/esp_now.html
http://www.espressif.com/sites/default/files/documentation/2c-esp8266_non_os_sdk_api_reference_en.pdf chapter 3.8

This is related to #176 (mesh networking) and #188 (generalized packet objects)

nickzoic, 2017-10-08:
The API is subject to change but I've made a start at https://github.com/nickzoic/micropython-esp32/tree/esp32-espnow

mcauser, 2017-12-03:
It seems the protocol can be used on both ESP8266 + ESP32

nickzoic, 2017-12-04:
Yeah, I'd like to get around to implementing it on
MicroPython/ESP8266 as well.
Don't tell anyone, but the unencrypted form is pretty easy to send and
receive from a Linux "monitor" device as well :-)The encrypted version may take a little more reverse engineering.

bboser, 2018-02-16:
Is there somewhere a working example? 
I got this to work alright if I first connect to a WiFi ap, but I suppose the whole point is that this should not be needed.

Without AP connection, espnow.send throws RuntimeError: ESP-Now Unknown Error 0x306c..

There is an example in the espressif docs (https://github.com/espressif/esp-idf/tree/master/examples/wifi/espnow)... but I thought I'd ask here first.

nickzoic, 2018-02-16:
I assume you mean from the micropython espnow branch?
I'll have to rewrite some of this stuff, and provide a better example,
but here's an example of broadcast based onhttp://nick.zoic.org/art/rocket-surgery-airborne-iot-telemetry-buzzconf/

import network w = network.WLAN(network.STA_IF) w.active(True)
w.config(protocol=network.MODE_LR)  from esp import espnow espnow.init()
espnow_bcast = b'\xFF' * 6 espnow.add_peer(w, espnow_bcast)  seq = 0
while True: msg = "Hello World %d" % seq espnow.send(espnow_bcast, msg)
time.sleep(0.004) seq += 1

Receiving is even simpler:

import network w = network.WLAN() w.active(True)
w.config(protocol=network.MODE_LR)  from esp import espnow espnow.init()
while True: msg = espnow.recv() if msg: print(msg[1])The important bits are:

* the interfaces have to be on the same mode and channel
* the interfaces have to be active
* try broadcast first
* once you have a mac address for the peer you can add them and
  unicast to them.
I'm going to rewrite it so that espnow.recv() is easier to use
asynchronously or as a socket ... it is just a proof-of-concept at
the moment.

Mbonea-Mjema, 2018-04-28:
Hey am new to micropython can you tell me how to get the esp-now working on my esp8266..that is how to install it and some few examples on it.. thanks

deshipu, 2018-05-07:
@Mbonea-Mjema it's not yet implemented for the esp8266.

eviscares, 2018-10-04:
Any hope of the pr that was closed a couple of days ago being merged soonish? Would love to avoid lora as another complication.

johncblacker, 2019-03-20:
I'd like to build this esp-now branch of mp.  What version of the ESP-IDE and xtensa toolchain do I need to do so?  I've tried finding the correct ESP-IDE based on the SUPHASH in the Makefile but couldn't find it.

johncblacker, 2019-03-20:
I tried building this today and got a ton of linker errors:
------------------------ partial log from build below --------------------------
/home/jobla/esp32-idf/esp-idf/components/esp32/lib/libphy.a(phy_chip_v7_ana.o): In function `phy_en_hw_set_freq':
/home/cff/gittree/chip7.1_phy/chip_7.1/board_code/app_test/pp/phy/phy_chip_v7_ana.c:1967: undefined reference to `phy_exit_critical'
/home/jobla/esp32-idf/esp-idf/components/esp32/lib/libphy.a(phy_chip_v7_ana.o): In function `phy_dis_hw_set_freq':
/home/cff/gittree/chip7.1_phy/chip_7.1/board_code/app_test/pp/phy/phy_chip_v7_ana.c:1967: undefined reference to `phy_enter_critical'
/home/jobla/esp32-idf/esp-idf/components/esp32/lib/libphy.a(phy_chip_v7_ana.o): In function `phy_force_wifi_chan':
/home/cff/gittree/chip7.1_phy/chip_7.1/board_code/app_test/pp/phy/phy_chip_v7_ana.c:1967: undefined reference to `phy_exit_critical'
/home/cff/gittree/chip7.1_phy/chip_7.1/board_code/app_test/pp/phy/phy_chip_v7_ana.c:1967: undefined reference to `phy_enter_critical'
/home/jobla/esp32-idf/esp-idf/components/esp32/lib/libphy.a(phy_chip_v7_ana.o): In function `ram_chip_i2c_writeReg':
/home/cff/gittree/chip7.1_phy/chip_7.1/board_code/app_test/pp/phy/phy_chip_v7_ana.c:1967: undefined reference to `phy_exit_critical'
/home/cff/gittree/chip7.1_phy/chip_7.1/board_code/app_test/pp/phy/phy_chip_v7_ana.c:1967: undefined reference to `phy_enter_critical'
/home/jobla/esp32-idf/esp-idf/components/esp32/lib/libphy.a(phy_chip_v7_ana.o): In function `ram_chip_i2c_readReg':
/home/cff/gittree/chip7.1_phy/chip_7.1/board_code/app_test/pp/phy/phy_chip_v7_ana.c:1967: undefined reference to `phy_exit_critical'
/home/cff/gittree/chip7.1_phy/chip_7.1/board_code/app_test/pp/phy/phy_chip_v7_ana.c:1967: undefined reference to `phy_enter_critical'
/home/jobla/esp32-idf/esp-idf/components/esp32/lib/libphy.a(phy_chip_v7_ana.o): In function `wifi_track_pll_cap':
/home/cff/gittree/chip7.1_phy/chip_7.1/board_code/app_test/pp/phy/phy_chip_v7_ana.c:1967: undefined reference to `phy_exit_critical'
/home/jobla/esp32-idf/esp-idf/components/esp32/lib/libphy.a(phy_chip_v7_ana.o): In function `btpwr_pll_track':
/home/cff/gittree/chip7.1_phy/chip_7.1/board_code/app_test/pp/phy/phy_chip_v7_ana.c:1967: undefined reference to `phy_enter_critical'
/home/jobla/esp32-idf/esp-idf/components/esp32/lib/libphy.a(phy_chip_v7_ana.o): In function `bt_track_tx_power':
/home/cff/gittree/chip7.1_phy/chip_7.1/board_code/app_test/pp/phy/phy_chip_v7_ana.c:1967: undefined reference to `phy_exit_critical'
/home/jobla/esp32-idf/esp-idf/components/esp32/lib/libphy.a(phy_chip_v7_ana.o): In function `chip_v7_set_chan_nomac':
/home/cff/gittree/chip7.1_phy/chip_7.1/board_code/app_test/pp/phy/phy_chip_v7_ana.c:707: undefined reference to `phy_enter_critical'
/home/cff/gittree/chip7.1_phy/chip_7.1/board_code/app_test/pp/phy/phy_chip_v7_ana.c:723: undefined reference to `phy_exit_critical'
/home/jobla/esp32-idf/esp-idf/components/esp32/lib/libphy.a(phy_chip_v7_cal.o): In function `ram_set_txcap_reg':
/home/cff/gittree/chip7.1_phy/chip_7.1/board_code/app_test/pp/phy/phy_chip_v7_cal.c:179: undefined reference to `phy_enter_critical'
/home/jobla/esp32-idf/esp-idf/components/esp32/lib/libphy.a(phy_chip_v7_cal.o): In function `tx_pwctrl_background':
/home/cff/gittree/chip7.1_phy/chip_7.1/board_code/app_test/pp/phy/phy_chip_v7_cal.c:179: undefined reference to `phy_exit_critical'
/home/jobla/esp32-idf/esp-idf/components/esp32/lib/libnet80211.a(ieee80211_api.o):(.text.esp_wifi_scan_get_ap_records+0x8): undefined reference to `esp_mesh_is_scan_allowed'
/home/jobla/esp32-idf/esp-idf/components/esp32/lib/libnet80211.a(ieee80211_api.o): In function `esp_wifi_scan_get_ap_records':
(.text.esp_wifi_scan_get_ap_records+0x10c): undefined reference to `esp_mesh_is_scan_allowed'
/home/jobla/esp32-idf/esp-idf/components/esp32/lib/libnet80211.a(ieee80211_hostap.o):(.text.ieee80211_hostap_send_beacon_process+0x0): undefined reference to `esp_mesh_quick_funcs'
/home/jobla/esp32-idf/esp-idf/components/esp32/lib/libnet80211.a(ieee80211_hostap.o):(.text.hostap_input+0x4c): undefined reference to `is_esp_mesh_assoc'
/home/jobla/esp32-idf/esp-idf/components/esp32/lib/libnet80211.a(ieee80211_hostap.o): In function `hostap_input':
(.text.hostap_input+0xdae): undefined reference to `is_esp_mesh_assoc'
/home/jobla/esp32-idf/esp-idf/components/esp32/lib/libnet80211.a(ieee80211_ioctl.o):(.text.wifi_mesh_ie_crypto_key_set_progress+0xc): undefined reference to `mesh_set_ie_crypto_config'
/home/jobla/esp32-idf/esp-idf/components/esp32/lib/libnet80211.a(ieee80211_ioctl.o): In function `wifi_mesh_ie_crypto_key_set_progress':
(.text.wifi_mesh_ie_crypto_key_set_progress+0x9f): undefined reference to `mesh_set_ie_crypto_config'
/home/jobla/esp32-idf/esp-idf/components/esp32/lib/libnet80211.a(ieee80211_ioctl.o):(.text.wifi_mesh_parent_candidate_set_progress+0x10): undefined reference to `mesh_set_parent_candidate'
/home/jobla/esp32-idf/esp-idf/components/esp32/lib/libnet80211.a(ieee80211_ioctl.o): In function `wifi_mesh_parent_candidate_set_progress':
(.text.wifi_mesh_parent_candidate_set_progress+0xb4): undefined reference to `mesh_set_parent_candidate'
/home/jobla/esp32-idf/esp-idf/components/esp32/lib/libnet80211.a(ieee80211_ioctl.o):(.text.wifi_mesh_parent_candidate_clear_progress+0x0): undefined reference to `mesh_clear_parent_candidate'
[... cut ...]
/home/jobla/esp32-idf/esp-idf/components/esp32/lib/libnet80211.a(ieee80211_mesh_quick.o): In function `ieee80211_mesh_quick_init':
(.text.ieee80211_mesh_quick_init+0x108): undefined reference to `ieee80211_init_mesh_assoc_ie'
Makefile:639: recipe for target 'build/application.elf' failed
make: *** [build/application.elf] Error 1
------------------------------- end of partial build log ----------------------------
Don't know exactly how to proceed at this point.  Hope someone has words of wisdom for me!

johncblacker, 2019-03-20:
I built MP from the esp-now repository and flashed to a couple of chips.  The send program seems to be OK, but the receiver gets an error that espnow.recv() doesn't exist.  Indeed if I do:
from esp import espnow
help(espnow)
I do not see a recv function.  Something is missing.  Perhaps there is a later repository that I should have built from?
Help would be greatly appreciated.

johncblacker, 2019-03-21:
I believe the receive example leaves a bit of tweaking.  First, there is no espnow.recv function, only a espnow.set_recv_cb(callback function);  secondly, I don't think the order of the statements is very helpful.  I tweaked it a bit to come up with:
import network
w = network.WLAN()
w.active(True)
w.config(protocol=network.MODE_LR)
from esp import espnow
espnow.init()
msg = ("", 0)

def receivecb(msg):
    print("in recv")
    if msg:
       print(msg[1])
        
espnow.set_recv_cb(receivecb(msg))

while True:
    pass
  #  msg = espnow.recv()

Problem is: I'm not getting anything printed out!
Anyone have any ideas?

johncblacker, 2019-03-21:
OK, fixed my received problem; here is the final send code (had an error) and receive code:
--------------------------- send code final version ----------------------------
import network 
import time
w = network.WLAN(network.STA_IF) 
w.active(True) 
w.config(protocol=network.MODE_LR) 
from esp import espnow 
espnow.init() 
espnow_bcast = b'\xFF' * 6
espnow.add_peer(w, espnow_bcast)
seq = 0
while True:
  msg = ("Hello World %d" % seq)
  espnow.send(espnow_bcast, msg)
  time.sleep(0.2)
  print("sent %d" % seq)
  seq += 1
--------------------------------- end of send code ------------------------------------------
and the receive code:
-------------------------------- start of receive code ----------------------------------------
import network
w = network.WLAN()
w.active(True)
w.config(protocol=network.MODE_LR)
from esp import espnow
espnow.init()
msg = ("", 0)

def receivecb(*dobj):
    print("in recv")
    if dobj:
      print(dobj[0])
      # print(" len, %d\n" % dobj[1])
        
espnow.set_recv_cb(receivecb)

while True:
    pass
  #  msg = espnow.recv()
----------------------------------------- end of receive code -----------------------------
Here is what receive prints out:
(b'0\xae\xa4\xfe\x92\x18', b'Hello World 89')

in recv

(b'0\xae\xa4\xfe\x92\x18', b'Hello World 90')

in recv

(b'0\xae\xa4\xfe\x92\x18', b'Hello World 91')

in recv

(b'0\xae\xa4\xfe\x92\x18', b'Hello World 92')

in recv

(b'0\xae\xa4\xfe\x92\x18', b'Hello World 93')

As you can see, the first member of the tuple is the mac address and the next part is the text send by the sender.  Since the sender is using a "broadcast" address ( = 6 bytes of 0x'ff') every device on the network would produce the same output.  A specific address can be used to send only to a specific machine.  But I haven't figured that out yet.

KKawase0104, 2019-03-26:
@johncblacker 

Hello.

Your script is very helpful.

When I run a script similar to your script for a long time, "core panic" is occured on the receiving side. Such a phenomenon was not happening?

Thank you.

https://github.com/micropython/micropython/pull/4115#issuecomment-476050962

johncblacker, 2019-03-27:
Sorry I can't help you; I didn't write the underlying espnow.c module, I just made changes to the originally posted script (which didn't work for me).  How long did it take before it fails?  Did you make your own firmware from the esp-now branch of the repository?  I only let my program run for a couple of minutes and it continued to work that long.  I believe the question would probably be best answered by the original author.  Remember, this is not officially released code; therefore, there may be (and probably are) bugs somewhere that will only surface with exhaustive testing.

KKawase0104, 2019-03-28:
Hello,

Thank you for your reply. I was using shawwwn's repository because I thought it was the newest esp now module, but now I see it's not your module, sorry about that.

commit 21e3426f7dec96c46d921fa16bb8e5db06572771

It took some time for the core panic to occur, but I don't remember exactly how long. When I tried after changing the sending side interval to 10msec, the core panic occurred after a couple minutes had passed. I also tried again using gc.mem_free on the receiving side. The timing of the core panic seems to be based on gc's collection timing.

The sending and receiving sides my original script were the same, which made it hard for me to tell where the core panic was occurring, but with your script I could see it was on the receiving side.

Thank you so much for your time.

nickzoic, 2020-01-21:
Discussion of this is now over at micropython/micropython#4115 and I'll make a new issue and/or pull request for on-going changes soon.

The problem you've described with panics is quite possibly caused by the memory allocation issue raised here: https://github.com/micropython/micropython/pull/4115#issuecomment-484375819

In the meantime I'm closing this issue as the micropython-esp32 repo is closed now.

closed 2020-01-21 by nickzoic