micropython/micropython-esp32 #202: time.mktime() and time.localtime() are missing

goatchurchprime, 2017-10-18:
These are a two purely numeric functions that do the calendar transformations between years/months/days to seconds since epoch.  They are in the ESP8266 library:

http://docs.micropython.org/en/latest/esp8266/library/utime.html?highlight=utime
```
MicroPython v1.9.1 on 2017-06-11; ESP module with ESP8266
Type "help()" for more information.
>>> import time
>>> dir(time)
['__name__', 'localtime', 'mktime', 'sleep', 'sleep_ms', 'sleep_us', 'ticks_ms', 'ticks_us', 'ticks_cpu', 'ticks_add', 'ticks_diff', 'time']
>>> 
```

But not the ESP32 library.

```
MicroPython v1.9.2-276-ga9517c04 on 2017-10-09; ESP32 module with ESP32
Type "help()" for more information.
>>> import time
>>> dir(time)
['__name__', 'time', 'sleep', 'sleep_ms', 'sleep_us', 'ticks_ms', 'ticks_us', 'ticks_cpu', 'ticks_add', 'ticks_diff']
>>> 
```

I can't account for how they are missing, as they are mentioned in the utime.rst docs:
    https://github.com/micropython/micropython-esp32/blob/046d15f074398cc2968fc66ae2fd25dc31100109/docs/library/utime.rst

I discovered this because the ESP32 machine module is missing the RTC class, but this is something I can easily work round if I have localtime() and mktime() functions instead.  

Have they been edited out of the utime library accidentally, or put somewhere else?  A more desperate work-around will be to find a pure python implementation of these two functions.

nickzoic, 2017-10-19:
Hi Julian,

This is just an oversight: the ESP32 port is still beta: it's very young
and is missing stuff that we just didn't think of yet.At this point the docs are aspirational.

I'm I'm submitting a PR now, thanks for bringing this to our attention.
-----N

dpgeorge, 2017-10-19:
Fixed by above PR.

closed 2017-10-19 by dpgeorge

goatchurchprime, 2017-10-19:
Dammit that was fast!  https://github.com/goatchurchprime/jupyter_micropython_developer_notebooks/blob/master/basicsockets/esptool.ipynb  Back to work.