micropython/micropython-esp32 #206: machine.reset() = Guru Meditation Error

diginfo, 2017-10-21:
Latest build, I am unable to capture the start of this crash as the first line below is repeated hundreds of time and is lost out of the screen buffer.

```
Guru Meditation Error of type IllegalInstruction occurred on core  0. Exception was unhandled.
Guru Meditation Error of type IllegalInstruction occurred on core  0. Exception was unhandled.
Guru Meditation E�ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
```

robert-hh, 2017-10-21:
if you are on Linux, you can 'tee' the output of the terminal emulator in a file, like:
`picocom -b 115200 /dev/ttyUSB0 | tee logfile`
Putty also has a log feature.

MrSurly, 2017-10-21:
Issue is discussed in https://github.com/micropython/micropython-esp32/issues/110 and a little bit in https://github.com/micropython/micropython-esp32/issues/204

closed 2019-01-16 by dpgeorge