micropython/micropython-esp32 #208: Second call to PWM constructor makes bad duty

goatchurchprime, 2017-10-25:
I hope I don't get a reputation for finding these obscure bugs, but I've reduced this down to the simplest example.  (And it doesn't manifest in the ESP8266).

It seems to get into a state where the duty number is invalid, and then issues system errors when you try to reset it.

```
>>> import machine
>>> p = machine.PWM(machine.Pin(2))
>>> p
PWM(2, freq=5000, duty=512)
>>> p = machine.PWM(machine.Pin(2))
>>> p
PWM(2, freq=5000, duty=16992)
>>> p.duty(40)
E (170024) ledc: ledc_set_duty(339): LEDC channel error
E (170024) ledc: ledc_update_duty(289): LEDC channel error
>>>
```

On another note, should the duty parameter just be doing [int_value]&0x03FF ?  It seems a bit confusing when it would be more intuitive if it (a) threw an exception if it's out of range, and (b) not throw an exception when I give it a float value.

MrSurly, 2017-10-25:
>I hope I don't get a reputation for finding these obscure bugs, but I've reduced this down to the simplest example.

That's a good reputation to have, IMHO.

dpgeorge, 2017-10-26:
Thanks for the report, should be fixed by b5385de6b25a2a79f9f69e0e1ef52c538a9b8880

closed 2017-10-26 by dpgeorge