micropython/micropython-esp32 #209: Socket timeout causes hard fault

BasedOnTechnology, 2017-10-26:
Somewhat related to #193, if a socket times out about 20 times the board hard faults.

```
Socket Open
Socket has timed out: [Errno 104] ECONNRESET
Socket Close

Socket Open
assertion "0" failed: file "../../py/../extmod/modussl_mbedtls.c", line 151, function: socket_new
abort() was called at PC 0x40131ad3 on core 0

Backtrace: 0x400873fb:0x3ffce020 0x40087427:0x3ffce040 0x40131ad3:0x3ffce060 0x400ecf2b:0x3ffce090 0x400df716:0x3ffce0e0 0x400dbbc1:0x3ffce110 0x400dbc29:0x3ffce130 0x400e8c09:0x3ffce150 0x400df7a4:0x3ffce1f0 0x400dbbc1:0x3ffce220 0x400e8b7f:0x3ffce240 0x400df7a4:0x3ffce2e0 0x400dbbc1:0x3ffce360 0x400e8b7f:0x3ffce380 0x400df7a4:0x3ffce420 0x400dbbc1:0x3ffce450 0x400dbbee:0x3ffce470 0x400f7352:0x3ffce490 0x400f75b5:0x3ffce530 0x400eeea7:0x3ffce550

Rebooting...
���x����xx�������
```
The relevant code:
```
def verify_server(socket):
    certHash = binascii.hexlify(
        hashlib.sha256(socket.getpeercert(True)).digest())
    print("Certificate Hash: " + str(certHash))
    for server in server_dict:
        if certHash in server['fingerprints']:
            return True
    return False

def send_to_server(data, url, server_index=0):
    wdt.feed()
    print("Sending " + str(data) + "to server " + url)
    s = socket.socket()
    print("Socket Open")
    try:
        s.settimeout(30)
        # server redundancy should be implemented. #FIXME: This will probably
        # do a dns lookup every time - probably shouldn't.
        ai = socket.getaddrinfo(server_dict[server_index]['address'], 443)
        addr = ai[0][-1]  # cache this
        s.connect(addr)
        s = ssl.wrap_socket(s, cert=sump_cert_2.CERT, key=sump_cert_2.KEY)

        if not verify_server(s):
            print("Certificate doesn't match.")
            s.close()
            return None

        server = server_dict[server_index]
        http_headers = '\r\n'.join([
            "POST {} HTTP/1.1".format(url),
            "Connection: Keep-Alive",
            "Accept: */*",
            "Host: {}".format(server['address']),
            "User-Agent: SafeSumpSeldon",
            "Content-Type: application/x-msgpack",
            "Content-Length: {}".format(len(data)),
            "\r\n"])
        s.write(http_headers)
        s.write(data)
        response = s.read(2048)
    except OSError as e:
        print("Socket has timed out: {}".format(e))
    except Exception as e:
        print("Server connection failed: {}".format(e))
        store_in_backlog(data)
    else:
        return response
        # process_backlog()
    finally:
        s.close()
        print("Socket Close")
    return None

```

We've been having numerous issues with sockets hardfaulting recently - it's a pretty serious issue for us.

Thanks!

MrSurly, 2017-10-26:
@danicampora That's usually a memory allocation error, if the mbedtls documentation is to be believed.

BasedOnTechnology, 2017-10-26:
@MrSurly The plot thickens.

Tried running 

```
for i in range(0,500):
    try:
        send_to_server()
    except:
        pass
```

If the server's off, I get:

```
Sending test to server /update
Socket Open
Socket has timed out: [Errno 104] ECONNRESET
OSError: 
```
Interesting that it never gets to s.close()? Weird. Anyhow, gc.mem_free() counts down by 0.5k every loop, and after about 60 cycles I get a string of OSErrors. Adding gc.collect() fixes that.

Even weirder, though, is what happens if the ssl handshake gets interrupted (server gets killed, wifi drops, etc):

```
Socket Open
mbedtls_ssl_handshake error: -68
Socket has timed out: [Errno 5] EIO
gc.mem_free(): 45552

...

Socket Open
Socket has timed out: [Errno 104] ECONNRESET
42000
Sending test to server 
Socket Open
assertion "0" failed: file "../../py/../extmod/modussl_mbedtls.c", line 151, function: socket_new
abort() was called at PC 0x40131a87 on core 0

Backtrace: 0x400873fb:0x3ffce0f0 0x40087427:0x3ffce110 0x40131a87:0x3ffce130 0x400ecf2b:0x3ffce160 0x400df716:0x3ffce1b0 0x400dbbc1:0x3ffce1e0 0x400dbc29:0x3ffce200 0x400e8c09:0x3ffce220 0x400df7a4:0x3ffce2c0 0x400dbbc1:0x3ffce2f0 0x400e8b7f:0x3ffce310 0x400df7a4:0x3ffce3b0 0x400dbbc1:0x3ffce430 0x400dbbee:0x3ffce450 0x400f7352:0x3ffce470 0x400f759c:0x3ffce510 0x400eeed8:0x3ffce550

Rebooting...

```

MrSurly, 2017-10-26:
> Interesting that it never gets to s.close()? Weird. Anyhow, gc.mem_free() counts down by 0.5k every loop, and after about 60 cycles I get a string of OSErrors. Adding gc.collect() fixes that.

These are all related; `close` de-allocates the memory associated with the mbedtls objects internal to the  μPy SSL object, but if it's never called, then there's a memory leak, but one that's outside of the MicroPython heap.

The  μPy garbage collector will destroy the  μPy SSL object, but the SSL objects don't have a finalizer.

There are two problems to be fixed:

* Figure out why `close()` is never called in your example.
* Add a finalizer to `modussl_mbedtls.c` for SSL objects that eventually hit the GC that haven't been `close`-ed properly.  `modsocket.c` has a finalizer for this purpose.

MrSurly, 2017-10-26:
I modified your code a bit so I could run it here, and it's working for me.  It errors out, but `s.close()` is being called.

```python
#!/usr/bin/env python3
import socket
import ssl
import binascii
import sys

def send_to_server(data, url, server_index=0):
    print("Sending " + str(data) + "to server " + url)
    s = socket.socket()
    print("Socket Open")
    address = '127.0.0.1'
    try:
        s.settimeout(30)
        # server redundancy should be implemented. #FIXME: This will probably
        # do a dns lookup every time - probably shouldn't.
        ai = socket.getaddrinfo(address, 443)
        addr = ai[0][-1]  # cache this
        s.connect(addr)
        s = ssl.wrap_socket(s, cert=sump_cert_2.CERT, key=sump_cert_2.KEY)

        http_headers = '\r\n'.join([
            "POST {} HTTP/1.1".format(url),
            "Connection: Keep-Alive",
            "Accept: */*",
            "Host: {}".format(address),
            "User-Agent: SafeSumpSeldon",
            "Content-Type: application/x-msgpack",
            "Content-Length: {}".format(len(data)),
            "\r\n"])
        s.write(http_headers)
        s.write(data)
        response = s.read(2048)
    except OSError as e:
        print("Socket has timed out: {}".format(e))
        sys.print_exception(e)
    except Exception as e:
        print("Server connection failed: {}".format(e))
        sys.print_exception(e)
    else:
        return response
        # process_backlog()
    finally:
        s.close()
        print("Socket Close")
    return None

def test():
    for i in range(0, 500):
        try:
            send_to_server('hello', 'strawberry.local')
        except Exception as e:
            print ('outer exception', e)
            sys.print_exception(e)
```

dpgeorge, 2017-10-26:
I've updated the ESP IDF to the latest version (note that it requires updating the xtensa toolchain), so the behaviour of this bug may have changed.

BasedOnTechnology, 2017-10-26:
@dpgeorge Yep:

I'm not getting that weird socket close behavior, all timeout mem-leaks are gone, however, if the SSL connection is interrupted for some reason, the thing still crashes:

```
Socket Open
Socket has timed out: [Errno 104] ECONNRESET
Socket Close
129
Sending test to server /update
Socket Open
assertion "0" failed: file "../../py/../extmod/modussl_mbedtls.c", line 151, function: socket_new
abort() was called at PC 0x401323ff on core 0

Rebooting...

```

 Interestingly, it's not unrecoverable - the esp completes rebooting and functions fine again. I think I'm okay with that behavior for now.

Thanks!

dpgeorge, 2017-10-27:
@BasedOnTechnology the assertion failing on line 151 looks like it can only happen due to an out-of-memory condition in mbedtls and is not related to the actual socket connection.  How exactly do you trigger this error?  Is it reproducible?

MrSurly, 2017-10-27:
@dpgeorge What do you think about adding finalizer to SSL sockets? Regular sockets already have one.  I think that that could definitely be one cause of this, since `close` for SSL is the only thing that releases mbedtls RAM, and it's not called during GC.

dpgeorge, 2017-10-27:
> What do you think about adding finalizer to SSL sockets?

Yes it makes sense to do that.

MrSurly, 2017-10-27:
I'll add it to mainline, though I don't have HW to test.

BasedOnTechnology, 2017-10-27:
@dpgeorge Yep, the error's completely reproducible. I have control over the server the board is connecting to - if I kill it (simulating a drop in wifi, say), the board always resets.

BasedOnTechnology, 2017-10-27:
Sorry to be constantly pasting dumps here, but here's the behavior I saw just now, with esp-idf et al up to date (might be a few red herrings in here, a bunch of code is called):

Timeline (server was off, might have come online for one or two requests):

- ECONNRESETs a bunch of times
- mbedtls_ssl_handshake error: -68
- EIO
- mbedtls_ssl_handshake error: -10
- EIO
- Station mode disconnects spontaneously?
- I (3884851) wifi: STA_DISCONNECTED, reason:201
- My softAP function runs
- SoftAP function connects to station mode again
- I kill and soft-reboot
- Tries to connect to server again
- Crashes instantly

```
Socket Open
Socket has timed out: [Errno 104] ECONNRESET
Socket Close

Socket Open
Socket has timed out: [Errno 104] ECONNRESET
Socket Close
Traceback (most recent call last):
  File "main.py", line 548, in <module>
  File "main.py", l
AttributeError: 'NoneType' object has no attribute 'decode'
<^ this is normal, my routines didn't get any data. I catch the error and continue.>

Socket Open
mbedtls_ssl_handshake error: -68
Socket has timed out: [Errno 5] EIO
Socket Close
Traceback (most recent call last):
  File "main.py", line 548, 
  File "main.py", line 525, in process_data
AttributeError: Type object has no attribute 'decode'

Socket Open
mbedtls_ssl_handshake error: -10
Socket has timed out: [Errno 5] EIO
Socket Close

I (3877511) wifi: bcn_timout,ap_probe_send_start
I (3880021) wifi: ap_probe_send over, resett wifi status to disai: state: run -> init (1)
I (3880021) wifi: n:5 0, o:5 0, ap:255 0, prof:1
I (3880021) wifi: pm stop, total sleep time: 0/-424

I (3880021) wifi: STA_DISCONNECTED, reason:200
bn timeout
no AP found
I (3884851) wifi: STA_DISCONNECTED, reason:201
no API (3887261) wifi: STA_DISCONNECTED, reason:201
no AP found
I (3889671) wifi: STA_DISCONNECTED, reason:201
no AP found
I (3890031) wifi: mode : sta (30:ae:a4:0d:a7:bc) + softAP (30:ae:a4:0d:a7:bd)
I (3890041) wifi: mode : softAP (30:ae:a4:0d:a7:bd)
I (3890041) wifi: event 13
I (3890041) wifi: event 3
I (3890041) wifi: event 1I (3890051) wifi: event 13
I (3890051) event: sta ip: 1.1.1.1, mask: 255.255.5.0, gw: 1.0.0.0
I (3890051) wifi: GOT_IP
I (3999521) event: station ip lost
I (3999521) wifi: event 8
i: GOT_IP
I (4090151) wifi: mode : sta (:a4:0d:a7:bc)
I (4090151) wifi: event 14

I (4584251) wifi: STA_DISCONNECTED, reason:201
no API (4584621) uart: queue free spaces: 10
I (4586661) wifi: STA_DISCONNECTED, reason:201
no API (4586781) wifi: n:5 0, o:1 0, ap:255 255, sta:5 0, prof:1
I (4586781) wifi: state: init -> auI (4586811) wifi: state: auth -> assoc (0)
I (4586811) wifi: state: assoc -> run (10)
I (4586961) wifi: connected with Arthur_Dent_Private_Network, channel 5
I (4586971) wifi: event 4

I (4588671) event: sta ip: 192.168.0.15, mask: 255.255.255.0, gw: 192.168.0.1
[0;32 GOT_IP
I (4588751) wifi: n:5 0, o:5 0, ap:255 255, sta:5 0, prof:1
I (588751) wifi: state: run -> auth (b0)
I (4589171) wifi: state: run -> auth (7a0)
I (4589171) wifi: n:5 0, o:5 0, ap:255 255, sta:5 0,:1
I (4589171) wifi: STA_DISCONNECTED, reason:7
I (4589291) wifi: n:5 0, o:5 0, ap:255 255, sta:5 0, prof:1
I (4589291) wifi: state: auth -> auI (4589311) wifi: state: auth -> assoc (0)
I (4589321) wifi: state: assoc -> run (10)
I (4592321) wifi: pm start, type:0

I (4598631) wifi: mode : sta (30:ae:a4:0d:a7:bc) + softAP (30:ae wifi: pm stop, total sleep time: 0/6310280

I (4598631) wifi:: run -> init (0)
I (4598631) wifi: n:5 0, o:5 0, ap:5 1, sta:5rof:1
I (4598631) wifi: mode : softAP (30:ae:a4:0d:a7:bd)
I (4598631) wifi: event 198641) wifi: event 3
I (4598641) event: sta ip: 1.1.sk: 255.255.255.0, gw: 1.0.0.0
I (4598641) wifi: GOT0m
I (4627351) wifi: mode : sta (30:ae:a4:0d:a7:bc) + softAP (30:ae:a4:0d:a7:bd)
I (4627351) wifi: mode : sta (30:ae:a4:0d:a7:bc)
Parsing msgpack
I (4627ifi: STA_START
I (4627351) wifi: event 14

<Here I hit Ctl-C Ctl-D>

Traceback (most recent call last):
KeyboardInterrupt: 

PYB: soft reboot

I (4634271) wifi: connected with Arthur_Dent_Private_Network, channel 5

Sending                                                                                                                                                           Socket Open
assertion "0" failed: file "../../py/../extmod/modussl_mbedtls.c", line 151, function: socket_neat PC 0x401323ff on core 0

Backtrace: 0x40087a8f:0x3ffce1a0 07abb:0x3ffce1c0 0x401323ff:0x3ffce1e0 0x400ecf9b:0x3ffce210 0x40ce260 0x400dbc31:0x3ffce290 0x400dbc99:0x3ffce2b0 0x400e8c79:0x30x400df814:0x3ffce370 0x400dbc31:0x3ffce3a0 0x400e8bef:0x3ffce3cf814:0x3ffce460 0x400dbc31:0x3ffce4e0 0x400e8bef:0x3ffce500 0x4014:0x3ffce5a0 0x400dbc31:0x3ffce5d0 0x400dbc5e:0x3ffce5f0 0x400fx3ffce610 0x400f76d9:0x3ffce6b0 0x400eef17:0x3ffce6d0

Rebooting...

```

That...is one sick ESP32. 

(apologies for any corruption in the dump, ground wasn't connected to my serial adapter)

Sorry for all the trouble guys.

BasedOnTechnology, 2017-10-27:
Okay, it happened again, much cleaner this time:
```

Sending 
Socket Open
mbedtls_ssl_handshake error: -7280
Socket has timed out: [Errno 5] EIO
Socket Close
Traceback (most recent call last):
  File "main.py", line 548, in <module>
  File "main.py", line 525, in process_data
Attrib 'NoneType' object has no attribute 'decode'

Sending 
Socket Open
mbedtls_ssl_handshake error: -10
Socket has timed out: [Errno 5] EIO
Socket Close
Traceback (most recent call last):
  File "main.py", line 548, in <module>
  File "main.py", line 525, in process_data
Attribrror: 'NoneType' object has no attribute 'decode'
I (2783481) wifi: bcn_timout,ap_probe_send_start
I (2785991) wifi: ap_probe_send over, resett wifi status to disassoc
I (2785991) wifi: state: r
                          I (2785991) wifi: n:5 0, o:5 0, ap:255 255, sta:5 0, prof:1
I 5991) wifi: pm stop, total sleep time: 0/-1518605307

I wifi: STA_DISCONNECTED, reason:200
beacon timeout
I (2788411) wifi: STA_DISCONNECTED, reason:201
no AP
     I (2790821) wifi: STA_DISCONNECTED, reason:201
no AP found
I (2793231) wifi: STA_DISCONNECTED, reason:201
no AP found
I (2795641) wifi: STA_DISCONNECTED, reason:201
no API (2796011) wifi: mode : sta (30:ae:a4:0d:a7:bc) + softAP (30:aebd)
I (2796011) wifi: mode : softAP (30:ae:a4:0d:a7:bd)
I (2796011) wifi: event 13
I (2796011) wifi: event 3
I (2796011) wifi: event 14
I (2796021) wifi: event 13
I (2796021) event: sta ip: 1.1.1.1, mask: 255.255. gw: 1.0.0.0
I (2796021) wifi: GOT_IP
I (2905511) event: station ip lost
I (2905511 wifi: event 8
I (2996131) wifi: mode : sta (30:ae:a4:0d:a7:bc) + softAP (30:ae2mI (2996131) wifi: STA_START
I (2996131) wifi: mode : sta :a4:0d:a7:bc)
I (2996131) wifi: event 14

Sending 
Socket Open
Socket has timed out: 118
Socket Close
Traceback (most recent call last):
  File "main.py", line 548, >
  File "main.py", line 525, in process_data
AttributeError: Type' object has no attribute 'decode'
I (2996431) wifi: n:5 0, o:1 0, ap:255 255, sta:5 0, prof:1
I (2996431) wifi: state: init -> auth (b0)
I (2996451) wifi: state: auth -> assoc (0)
I (2996461) wifi: state: assoc -> run (10)
I (2996581) wifi: connected with Arthur_Dent_Private_Network, channel 5
I (2996581) wifi: event 4
I (2998641) event: sta ip: 192.168.0.15, mask: 255.255.252.168.0.1
I (2998641) wifi: GOT_IP

I (2999461) wifi: pm start, type:0

Sending 
Socket Open
assertion "0" failed: file "../../py/../extmod/modussl_mbedtls.c", line 151, function: socket_ne() was called at PC 0x401323ff on core 0

Backtrace: 0x40087a8ce1a0 0x40087abb:0x3ffce1c0 0x401323ff:0x3ffce1e0 0x400ecf9b:0x310 0x400df786:0x3ffce260 0x400dbc31:0x3ffce290 0x400dbc99:0x3ffc0x400e8c79:0x3ffce2d0 0x400df814:0x3ffce370 0x400dbc31:0x3ffce3a00e8bef:0x3ffce3c0 0x400df814:0x3ffce460 0x400dbc31:0x3ffce4e0 0bef:0x3ffce500 0x400df814:0x3ffce5a0 0x400dbc31:0x3ffce5d0 0x400:0x3ffce5f0 0x400f7476:0x3ffce610 0x400f76d9:0x3ffce6b0 0x400eef3ffce6d0

Rebooting...
I (348) cpu_start: Pro cpu up.
I (348) cpu_st[0m
I (348) heap_init: Initializing. RAM available for dc allocation:
I (349) heap_init: At 3FFAE6E0 len 000 (6 KiB): DRAM
I (350) heap_init: At 3FFD4CE0 len 00 (44 KiB): DRAM
I (352) heap_init: At 3FFE0440 len 0BC0 (14 KiB): D/IRAM
I (353) heap_init: At 3FFE4350 001BCB0 (111 KiB): D/IRAM
I (355) heap_init: At 4008en 00010500 (65 KiB): IRAM
I (356) cpu_start: Pro cper code
I (143) cpu_start: Starting scheduler on PRO CPU.
I (1249) wifi: wifi firmware version: 111e74d
I (1249) wifi: co (1249) wifi: config nano formating: disabled
I (1249) sBase MAC address is not set, read default base MAC address from of EFUSE
I (1249) system_api: Base MAC address is no, read default base MAC address from BLK0 of EFUSE
I (1259) wifi: Init dynamic tx buffer num: 32
I (1259) wifi: Init data frame dynamic rx buffer 9) wifi: Init management frame dynamic rx buffer num: 64
I (125i: wifi driver task: 3ffdf440, prio:23, stack:4096
I (1259) wiftic rx buffer num: 10
I (1269) wifi: Init dynamic rx buffer num69) wifi: Init rx ampdu len mblock:7
I (1269) wifi: Init lldescpdu entry mblock:4
I (1269) wifi: wifi power manager task: 0x3f0 prio: 21 stack: 2560
I (1289) phy: phy_version: 362.0, 61e8d92, Sep  8 2017, 1289) wifi: mode : null
I (1299) modsocket: Initializing
I (1419) uart: ALREADY NULL
I (1419) uart: queue free spaces: 10
b''
I (5429) wifi: mode : sta (30:ae:a4:0d:a7:bc)
I (5429) wifi: STA_START
I (6039) wifi: n:5 0, o:1 0, ap:255 255, sta:5 0, prof:1
I (6599) wifi: state: init -> auth (b0)
I (6619) wifi: state: auth -> assoc (0)
I (6629) wifi: state: assoc -> run (10)
I (6789) wifi: connected with Arthur_Dent_Private_Network, channI (6789) wifi: event 4
I wifi: GOT_IP
I (9629) wifi: pm start, type:0

Parsing msgpack
Parsing HDLC - input data: 

Sending r /update
Socket OpCertificate Hash: b'566ec9a1e45fcfceb43f7b82204076fcfb535506d99e63a089938'
Socket Close

Socket Open
Certificate Hash: b'566ec9a1e45fcfceb43f7b82204076fcfb535506d99ee2bd84e15de63a089938'
Socket Close
```

dpgeorge, 2017-10-30:
Finaliser support for ussl objects was added in a6566fc2e15e8efd856b39c7ee2132fcdad3450c

BasedOnTechnology, 2017-10-30:
Sorry guys, but that didn't seem to do it.

```
Socket Open
mbedtls_ssl_handshake error: -7280
Socket has timed out: [Errno 5] EIO
Socket Close

Socket Open
mbedtls_ssl_handshake error: -68
Socket Close

Socket Open
mbedtls_ssl_handshake error: -10
Socket has timed out: [Errno 5] EIO
Socket Close
Traceback (most recent call last):
  File "main.py", line 550, in <module>
  File "main.py", line 527, in process_data
AttributeError: 'NoneType' object has no attribute 'decode'
I (303267) wifi: bcn_timout,ap_probe_send_start
I (305767) wifi: ap_probe_send over, resett wifi status to disassoc
I (305767) wifi: state: run -> init (1)
I (305767) wifi: n:5 0, o:5 0, ap:255 255, sta:5 0, prof:1
I (305767) wifi: pm stop, total sleep time: 0/297598933

I (305767) wifi: STA_DISCONNECTED, reason:200
beacon timeout
I (308187) wifi: STA_DISCONNECTED, reason:201
no AP found
I (310597) wifi: STA_DISCONNECTED, reason:201
no AP found
I (313007) wifi: STA_DISCONNECTED, reason:201
no AP found
I (315417) wifi: STA_DISCONNECTED, reason:201
no AP found

Socket Open
mbedtls_ssl_handshake error: -4290
Socket has timed out: [Errno 5] EIO
Socket Close

Socket Open
Socket has timed out: 118
Socket Close

```

BasedOnTechnology, 2017-10-30:
By varying the timing at which I kill the server / number of times I repeat the handshake abort, I can get it to display different behaviour:
```
Socket Open
mbedtls_ssl_handshake error: -10
Socket has timed out: [Errno 5] EIO
Socket Close
Traceback (most recent call last):
  File "main.py", line 548, in <module>
  File "main.py", locess_data
AttributeError: 'NoneType' object has no attribute ''
Parsing msgpack
Parsing HDLC - input data: b'\x93\x03\xcdR2\xde\x00\x11\xaawaterlevel\xcaA\x0f\x81\xa0\xafbatteracwall_voltage\xcaAp\xcc\xa8\xa9pump_duty\x92\xcaB\x0f\x9d\x11\x\x00\x00\x00\xb3flowrate_calculated\xca\x00\x00\x00\x00\xb1flowrbserved\xca\x00\x00\x00\x00\xa6cycles\x00\xadstd_deviation\xca?\\xacpump_current\x92\xca\x00\x00\x00\x00\xca\x00\x00\x00\x00\xaee_current\xca>#\x8b\x8b\xabcharge_duty\xca?\x80\x00\x00\xacraw_sE\x7f\x93J\xa4temp\xca\xc2\x9b\xb3\xe9\xaepump_deviation\x92\xca8\xe3\xca\xbfFW\xef\xaecharge_current\xca>#\x8b\x8b\xabcharge_du?\x80\x00\x00\xa7version\xca?\x99\x99\x9a~'

Output hdlc: b'\x93\x03\xcdR2\xde\x00\x11\xaawataA\x0f\x81\xa0\xafbattery_voltage\xca=G\xa6f\xacwall_voltage\xcaxa9pump_duty\x92\xcaB\x0f\x9d\x11\xca\x00\x00\x00\x00\xb3flowratted\xca\x00\x00\x00\x00\xb1flowrate_observed\xca\x00\x00\x00\x00les\x00\xadstd_deviation\xca?\x81\x1e\xda\xacpump_current\x92\xc\x00\x00\x00\xca\x00\x00\x00\x00\xaecharge_current\xca>#\x8b\x8bharge_duty\xca?\x80\x00\x00\xacraw_salinity\xcaE\x7f\x93J\xa4tem\xc2\x9b\xb3\xe9\xaepump_deviation\x92\xca\xbf\xfe\x08\xe3\xca\xef\xaecharge_current\xca>#\x8b\x8b\xabcharge_duty\xca?\x80\x00\xa7version\xca?\x99\x99\x9a'
Parsed to [3, 21042, {'raw_salinity': 4089.206, 'charge_duty': 1.0, 'flowrate_calculated': 0.0, 'temp': -77.85139, 'cycles': 0, ump_deviation': [-1.984646, -0.7747793], 'version': 1.2, 'charge0.159712, 'std_deviation': 1.008754, 'waterlevel': 8.969147, 'ba_voltage': 0.04874268, 'pump_duty': [35.90338, 0.0], 'flowrate_o.0, 'pump_current': [0.0, 0.0], 'wall_voltage': 15.04997}]
MessSAM_UPDATE
Sending b'\x93\x03\xcdR2\xde\x00\x11\xaawaterlevel\xcaA\x0f\x81\tage\xca=G\xa6f\xacwall_voltage\xcaAp\xcc\xa8\xa9pump_duty\x92\xf\x9d\x11\xca\x00\x00\x00\x00\xb3flowrate_calculated\xca\x00\x00x00\xb1flowrate_observed\xca\x00\x00\x00\x00\xa6cycles\x00\xadstiation\xca?\x81\x1e\xda\xacpump_current\x92\xca\x00\x00\x00\x00\00\x00\x00\x00\xaecharge_current\xca>#\x8b\x8b\xabcharge_duty\xc0\x00\x00\xacraw_salinity\xcaE\x7f\x93J\xa4temp\xca\xc2\x9b\xb3\aepump_deviation\x92\xca\xbf\xfe\x08\xe3\xca\xbfFW\xef\xaechargeent\xca>#\x8b\x8b\xabcharge_duty\xca?\x80\x00\x00\xa7version\xcax99\x9a'to server /update
Socket Open
assertion "0" failed: file "../../py/../extmod/modussl_mbedtls.ce 151, function: socket_new
abort() was called at PC 0x401323ffre 0

Backtrace: 0x40087a8f:0x3ffce1a0 0x40087abb:0x3ffce1c0 023ff:0x3ffce1e0 0x400ecf9b:0x3ffce210 0x400df786:0x3ffce260 0x401:0x3ffce290 0x400dbc99:0x3ffce2b0 0x400e8c79:0x3ffce2d0 0x400dfx3ffce370 0x400dbc31:0x3ffce3a0 0x400e8bef:0x3ffce3c0 0x400df814fce460 0x400dbc31:0x3ffce4e0 0x400e8bef:0x3ffce500 0x400df814:0x5a0 0x400dbc31:0x3ffce5d0 0x400dbc5e:0x3ffce5f0 0x400f7476:0x3ff76d9:0x3ffce6b0 0x400eef17:0x3ffce6d0

```

BasedOnTechnology, 2017-10-30:
Here's the bare minimum code that still reproduces this issue. 

Make an ssl server with 
`openssl req -x509 -newkey rsa:2048 -keyout key.pem -out cert.pem -days 365 -nodes`
`openssl s_server -key key.pem -cert cert.pem -accept 0.0.0.0:8000 -www`

With the latest daily build from micropython.org, run

```
import network
import socket
import ussl
import utime
wlan = network.WLAN(network.STA_IF)
wlan.active(True)
wlan.connect(wifinetwork, password)
utime.sleep(10)
while(True):
	try:
		s = socket.socket()
		s.settimeout(30)
		ai = socket.getaddrinfo('192.168.0.42', 8000)
		addr = ai[0][-1]
		s.connect(addr)
		s = ussl.wrap_socket(s)
		s.close()
	except:
		print('Timeout')

```
If you kill the server with Ctrl-C, disconnect wifi, or even just leave it running, you'll see a bunch of different behaviors. 

Thanks!

BasedOnTechnology, 2017-10-31:
Adding gc.collect() seems to completely solve the issue now.

MrSurly, 2017-10-31:
> Adding gc.collect() seems to completely solve the issue now.

That's odd.  I've seen that before, but only with the Pycom fork.

dpgeorge, 2017-10-31:
I think I found the issue: there's only enough RAM (on the FreeRTOS heap) to create 2 SSL connections.  If you try to create a third while the others are active then it will hit an assertion because mbedtls can't allocate any more memory.

And this case is being hit in the above test by the following sequence of events:
1. server dies during a handshake, leading to a memory leak on esp32
2. server restarts, connections can be made
3. server dies again during a handshake, leading to another memory leak
4. server restarts
5. next connection fails due to out-of-memory

The fix (applied temporarily in this repo in 54c6ebcfd72cae2b3000987f99651e9c8015bf18, @BasedOnTechnology please can you test this fix?) is to clean up the mbedtls state when there is an error during the creation of the ssl object.

BasedOnTechnology, 2017-11-03:
@dpgeorge Thanks for Hi, I haven't encountered any issues with this fix. Tentatively, it looks like it's solved the problem.