micropython/micropython-esp32 #212: OS Error 118 on sock.connect(addr)

PikWay, 2017-10-29:
I'm trying to do simple socket conn and on sock.connect(addr) I'm getting OS Error 118. I really don't know what this means because it's absent in uerrno. 
 
ESP is running on firmware: esp32-20171029-v1.9.2-283-gdba3796e

PikWay, 2017-10-29:
I've just found that wlan.isconnected() is always True so it's not usocket issue. I was using my ESP8266 wifi script which seems to not working on ESP32.

MrSurly, 2017-10-30:
Can you post the full code? Or at least enough to reproduce the issue?

nickzoic, 2017-10-30:
I suspect this is related to https://github.com/micropython/micropython-esp32/issues/166#issuecomment-328731393 with the inconsistent errno definitions ...

PikWay, 2017-10-30:
import network
wifi_sta = network.WLAN(network.STA_IF)
wifi_sta.isconnected()
False
wifi_sta.active(True)
I (157116) wifi: mode : sta (<masked MAC>)
I (157116) wifi: STA_START
True
wifi_sta.isconnected()
False
wifi_sta.ifconfig(('192.168.0.105','255.255.255.0','192.168.0.50','8.8.8.8'))
wifi_sta.isconnected()
True 

I'm not connected yet. In ESP8266 in this stage it's showing False so i was entering following if:
if not wifi_sta.isconnected()

And making the connection. This OS Error 118 is effect of above - cannot connect to socket while not connected to network.

Anyway I've moved wifi_sta.ifconfig() into "if" and it works.

PikWay, 2017-10-30:
and following code will throw OS Error 118:

import usocket as socket
I (1112246) modsocket: Initializing
addr=socket.getaddrinfo('192.168.0.66',31883)[0][-1]
sock=socket.socket()
sock.connect(addr)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: 118

dpgeorge, 2017-10-31:
@PikWay your report here is not 100% clear on what your issue is.  The above code with the "OSError: 118" is simply due to the network/wifi not being connected.

> This OS Error 118 is effect of above - cannot connect to socket while not connected to network.

Yes, you must be connected to wifi for the socket to connect to a remote machine.  What else do you expect it to do?

PikWay, 2017-10-31:
@dpgeorge Ok. Will try to make it clearer. Agree that if trying to bind to socket on disconnected network should throw error. 

Problem is with determining is network connected when using static IP. 
In esp_isconnected there's just check:
return mp_obj_new_bool(info.ip.addr != 0);

So if network.WLAN.ifconfig was invoked with addressing tuple - network.WLAN.isconnected will return True before connecting to WiFi. 
Is there any other way of determining connection status when using static IP

robert-hh, 2017-10-31:
The problem is, that in contrast to the ESP8266 port there is no dedicated API telling whether the device is connected. So other indications must be used, like esp_wifi_sta_get_ap_info(). That seems to work, but it is unclear to me whether this can fail even if a device is connected. Anyway, the function could could look like:
```
STATIC mp_obj_t esp_isconnected(mp_obj_t self_in) {
    wlan_if_obj_t *self = MP_OBJ_TO_PTR(self_in);
    if (self->if_id == WIFI_IF_STA) {
        wifi_ap_record_t ap_info;
        ap_info.primary = 0;
        return mp_obj_new_bool(esp_wifi_sta_get_ap_info(&ap_info) == ESP_OK && ap_info.primary != 0);
    } else {
        wifi_sta_list_t sta;
        esp_wifi_ap_get_sta_list(&sta);
        return mp_obj_new_bool(sta.num != 0);
    }
}
STATIC MP_DEFINE_CONST_FUN_OBJ_1(esp_isconnected_obj, esp_isconnected);
```

robert-hh, 2017-10-31:
The option above is only a little bit better. It will not flag "connected" after setting the IP adress. But if after connecting the connection is lost, it will still indicate an exiting connection.

PikWay, 2017-10-31:
Why not to use this variable: static bool wifi_sta_connected = false;?
Looks like it's set properly in both - esp_connect and esp_disconnect functions. 

Also I can see some events in logs, set to different values during wifi connection lifecycle.
I'll try to change this myself.

robert-hh, 2017-10-31:
That's more simple but fails too, because it will tell you immediately that a connection was made, before it is established, even if it gets never done.

robert-hh, 2017-10-31:
OK. Found a solution which seems solid. 
Adding a separate status variable, which is set True on a connect event and False on a disconnect event.
I stumbled over another glitch, in that disconnect() would do that, but it triggers a disconnect event, which immediately tries to re-establish the connection.

PikWay, 2017-11-09:
Robert solution works. Issue to close.

robert-hh, 2017-11-09:
You can close it yourself. Just push the 'close' button.

closed 2017-11-09 by PikWay