micropython/micropython-esp32 #214: dns address of the access point set to '0.0.0.0'

lemariva, 2017-10-30:
Using the following code to configure an acess point on the 'ESP32 module with ESP32', it results in a configuration of the access point with dns ip '0.0.0.0'.
```
>>> import network
>>> ap_if = network.WLAN(network.AP_IF)
>>> ap_if.active(True)
>>> ap_if.config(essid="network-name", authmode=network.AUTH_OPEN)
>>> ap_if.ifconfig()
('192.168.4.1', '255.255.255.0', '192.168.4.1', '0.0.0.0')
```
There is no possibility to configure the dns address. I am now writing a captive portal, that is why I need the dns to be configured at least with the gateway address. 
Using MicroPython on the WiPy 2.0 works. The WiPy2.0 has the following function:
```
>>> ifconfig(config=('192.168.4.1', '255.255.255.0', '192.168.4.1', '192.168.4.1'))
```
which is not available on the ESP32.

robert-hh, 2017-10-30:
This has been fixed last night: https://github.com/micropython/micropython-esp32/issues/210
At least tomorrow's daily build should work.

closed 2017-10-30 by lemariva

jczic, 2017-11-13:
That's doesn't works on last WiPy (3.0) firmware...?
The DNS address of the WLAN Access Point is always forced to 0.0.0.0.

MrSurly, 2017-11-13:
@jczic I believe the firmware for the WiPy is maintained by PyCom.