micropython/micropython-esp32 #215: Add support to ure for named-groups

abencomo, 2017-10-31:
Will it be possible to add support for named-groups in the ure api?  It will help a lot with the parsing of dynamic URIs and URLs containing named gapturing groups.

https://docs.python.org/3/howto/regex.html#non-capturing-and-named-groups

nickzoic, 2017-10-31:
G'day!  This would be a change to `extmod/modure.c` so it'd have to happen in the main micropython repo ...
https://github.com/micropython/micropython ...  

I agree they're a lot nicer to deal in some circumstances with but it might be that they'd make the re implementation a bit heavier than is desired on small platforms ...

abencomo, 2017-10-31:
Closing this one and opening a new one against the main repo as nickzoic pointed out.

https://github.com/micropython/micropython/issues/3403

closed 2017-10-31 by abencomo