micropython/micropython-esp32 #231: Connecting to AP that is "out-of-range"

PikWay, 2017-11-29:
Hi 
When I'm trying to connect to AP that is currently out-of-range ESP starts to spin trying to connect (with error: No AP found). Only way to stop it is reboot (until you don't have this in boot.py).
I've made workaround in my code, but can someone check is this expected behavior? Maybe n-attempts and then throw some exception?

nickzoic, 2017-11-29:
Have you got some example code?
Normally, you need to do something like:

wlan = network.WLAN()
wlan.connect('foo','bar')
while not wlan.isconnected():
    time.sleep(1)
print(wlan.ifconfig())

... because execution continues without waiting for the connection
to complete.

PikWay, 2017-11-29:
Code is just like yours:
import time,network
def connectWLAN():
	sta=network.WLAN(network.STA_IF)
	sta.connect('foo','bar')
	while not sta.isconnected():
		time.sleep(1)
	print(sta.ifconfig())

and after executing connectWLAN() it starts to  write to serial console:
I (118715) wifi: STA_DISCONNECTED, reason:201
no AP found
I (121545) wifi: STA_DISCONNECTED, reason:201
no AP found
I (124375) wifi: STA_DISCONNECTED, reason:201
no AP found
I (127205) wifi: STA_DISCONNECTED, reason:201
no AP found

Keyboart interrupt doesnt affect this, because it's trying to connect somewhere in background, meanwhile you cannot issue any commands in console (and sometimes ampy also doesn't work). I think it's related to #165 - I'll check that.

PikWay, 2017-11-29:
Checked dpgeorge fix, but seems like except to logger it prints "no AP found" 

>>> sta.connect('foo','bar')
>>> no AP found
no AP found
no AP found
no AP found

>>> no AP found

>>>
>>> no AP found
no AP found
no AP found
no AP found

>>>
>>>
>>>
>>> no AP found
no AP found
no AP found
no AP found
no AP found
no AP found
no AP found
no AP found

robert-hh, 2017-11-29:
Don't know if that helps, but you can stop those connect attempts and the related messages with disconnect(), e.g. after your timeout,  and try maybe later.

PikWay, 2017-11-29:
I've already made workaround. First scan, then check is essid I want to connect in list and if yes - connect.