micropython/micropython-esp32 #234: OSError: [Errno 1] EPERM

Lisa999, 2017-12-07:
esptool.py --port /dev/ttyUSB3 write_flash --compress 0x1000 build/firmware.bin
esptool.py v2.1
Connecting.....
Detecting chip type... ESP32
Chip is ESP32D0WDQ6 (revision 1)
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 1091056 bytes to 645654...
Wrote 1091056 bytes (645654 compressed) at 0x00001000 in 56.8 seconds (effective 153.8 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting...

>>> f = open("main.py", "w")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 1] EPERM
>>> 

I'm lost... It suspect it's the firmware size causing this, since it's now bigger then 1MB (8Mb). Help?

Avi-TelnT, 2017-12-07:
Did you run ./BUILD.sh menuconfig ? 
Select the right serial ? 
Enable Spiff file system?
Do you run Ubuntu?

Lisa999, 2017-12-07:
Did you run ./BUILD.sh menuconfig ?
- Why should i run this?
Select the right serial ?
- 'Wrote 1091056 bytes', so i guess: yes
Enable Spiff file system?
- Where? In the ESP32?
Do you run Ubuntu?
- To build the firmware and run esptool? Yes

robert-hh, 2017-12-07:
The 1 MB should not be the problem. The file system starts at 2 MB. You could try the usual crowbar approach, which is erasing flash and re-flashing the firmware. The error looks as if the file system is corrupted. After erasing, that will be rebuilt.

Avi-TelnT, 2017-12-07:
run ./BUILD.sh menuconfig
and enable Spiff file system
MicroPython |  File systems | Use SPIFFS
Save and rebuild all by /BUILD.sh all
and them do: /BUILD.sh flashfs

robert-hh, 2017-12-07:
@Avi-TelnT I may be, that @Lisa999 is NOT using the loboris branch and NOT using the plain ESP32 idf method. In that case, she/he cannot follow your suggestion.

Avi-TelnT, 2017-12-07:
You are right.... my bad.

MrSurly, 2017-12-07:
> The 1 MB should not be the problem. The file system starts at 2 MB. You could try the usual crowbar approach, which is erasing flash and re-flashing the firmware. The error looks as if the file system is corrupted. After erasing, that will be rebuilt.

I agree with @robert-hh  -- try doing a full erase first.

Also, the partitions were increased to account for the firmware size, a while back.

Lisa999, 2017-12-07:
@Avi-TelnT: Why would i put the error here, if i was using the loboris branch?

Lisa999, 2017-12-07:
**Make clean
Make
esptool.py --port /dev/ttyUSB3 erase_flash
esptool.py --port /dev/ttyUSB3 write_flash --compress 0x1000 build/firmware.bin**
esptool.py v2.1
Connecting....
Detecting chip type... ESP32
Chip is ESP32D0WDQ6 (revision 1)
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 1091056 bytes to 645654...
Wrote 1091056 bytes (645654 compressed) at 0x00001000 in 56.8 seconds (effective 153.8 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting...
**>>>f = open("main.py", "w")**
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 1] EPERM

:-(

robert-hh, 2017-12-07:
What board are you using?

MrSurly, 2017-12-07:
This is the command line I'm using.  I usually just do `make  -j4 PORT=/dev/ttyUSB2 deploy`.

    esptool.py --chip esp32 --port /dev/ttyUSB2 --baud 460800 write_flash -z --flash_mode dio --flash_freq 40m 0x1000 build/firmware.bin

Also, could you run `git log | head` and paste the output?

Lisa999, 2017-12-07:
Official DOIT ESP32 Development Board WiFi+Bluetooth Ultra-Low Power Consumption Dual Core ESP-32 ESP-32S ESP 32 Similar ESP8266
[https://nl.aliexpress.com/item/Official-DOIT-ESP32-Development-Board-WiFi-Bluetooth-Ultra-Low-Power-Consumption-Dual-Core-ESP-32-ESP/32801621054.html?isOrigTitle=true](url)

Lisa999, 2017-12-07:
commit d0678a3e56cdc4ee9fbb15e0f1d483dddf02f7bb
Merge: 2305d84 8a3e39f
Author: Lisa <Lisa999>
Date:   Sun Oct 22 17:36:13 2017 +0200

    Merge branch 'dev-deepsleep' of https://github.com/MrSurly/micropython-esp32 into esp32

commit 2305d84844bf77a92d161aae77475a1d1a87101f
Author: Nick Moore <nick@zoic.org>
Date:   Thu Oct 19 11:01:53 2017 +1100

Lisa999, 2017-12-07:
**esptool.py --chip esp32 --port /dev/ttyUSB3 --baud 460800 write_flash -z --flash_mode dio --flash_freq 40m 0x1000 build/firmware.bin**
esptool.py v2.1
Connecting....
Chip is ESP32D0WDQ6 (revision 1)
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 1091056 bytes to 645654...
Wrote 1091056 bytes (645654 compressed) at 0x00001000 in 15.9 seconds (effective 549.7 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting...
**>>> f = open("main.py", "w")**
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 1] EPERM

Broken? buy a new one?

MrSurly, 2017-12-07:
Go back to the basics:

* Do a checkout of _this_ repo (not my fork).  
* Make sure you don't get any ESP-IDF hash mismatches when you compile -- there will be a warning at the very beginning of the make process if they don't match.  Doing a `make clean` is an easy way to see if this error shows up.
* Before you build, do the following:
    * cd {idf repo}
    * git checkout {proper hash}
    * git submodule update --init --recursive
    * cd {micropython repo}
    * git submodule update --init --recursive
* make erase
* make clean
* make deploy

Lisa999, 2017-12-07:
This repo? Which one?

robert-hh, 2017-12-07:
The one we are in now.

Lisa999, 2017-12-07:
**After make clean, make, no warnings:**
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
mkdir -p build/bootloader//home/lisa/Programming/esp32/esp-idf/components/bootloader/subproject/main/
mkdir -p build/bootloader//home/lisa/Programming/esp32/esp-idf/components/bootloader_support/src/
mkdir -p build/bootloader//home/lisa/Programming/esp32/esp-idf/components/log/
mkdir -p build/bootloader//home/lisa/Programming/esp32/esp-idf/components/micro-ecc/micro-ecc/
mkdir -p build/bootloader//home/lisa/Programming/esp32/esp-idf/components/soc/esp32/
mkdir -p build/bootloader//home/lisa/Programming/esp32/esp-idf/components/spi_flash/
CC /home/lisa/Programming/esp32/esp-idf/components/bootloader_support/src/bootloader_flash.c
CC /home/lisa/Programming/esp32/esp-idf/components/bootloader_support/src/bootloader_random.c
CC /home/lisa/Programming/esp32/esp-idf/components/bootloader_support/src/bootloader_sha.c
CC /home/lisa/Programming/esp32/esp-idf/components/bootloader_support/src/secure_boot_signatures.c
CC /home/lisa/Programming/esp32/esp-idf/components/bootloader_support/src/secure_boot.c
CC /home/lisa/Programming/esp32/esp-idf/components/bootloader_support/src/esp_image_format.c
CC /home/lisa/Programming/esp32/esp-idf/components/bootloader_support/src/flash_encrypt.c
CC /home/lisa/Programming/esp32/esp-idf/components/bootloader_support/src/flash_partitions.c
CC /home/lisa/Programming/esp32/esp-idf/components/log/log.c
CC /home/lisa/Programming/esp32/esp-idf/components/spi_flash/spi_flash_rom_patch.c
CC /home/lisa/Programming/esp32/esp-idf/components/soc/esp32/rtc_clk.c
CC /home/lisa/Programming/esp32/esp-idf/components/soc/esp32/rtc_time.c
CC /home/lisa/Programming/esp32/esp-idf/components/micro-ecc/micro-ecc/uECC.c
CC /home/lisa/Programming/esp32/esp-idf/components/bootloader/subproject/main/bootloader_start.c
LINK build/bootloader.elf
Create build/bootloader.bin
esptool.py v2.1
Create build/partitions.bin
mkdir -p build/genhdr
Generating build/genhdr/mpversion.h
GEN build/genhdr/qstr.i.last
GEN build/genhdr/qstr.split
GEN build/genhdr/qstrdefs.collected.h
QSTR updated
GEN build/genhdr/qstrdefs.generated.h
mkdir -p build//home/lisa/Programming/esp32/esp-idf/components/app_trace/
mkdir -p build//home/lisa/Programming/esp32/esp-idf/components/cxx/
mkdir -p build//home/lisa/Programming/esp32/esp-idf/components/driver/
mkdir -p build//home/lisa/Programming/esp32/esp-idf/components/esp32/
mkdir -p build//home/lisa/Programming/esp32/esp-idf/components/esp32/hwcrypto/
mkdir -p build//home/lisa/Programming/esp32/esp-idf/components/ethernet/
mkdir -p build//home/lisa/Programming/esp32/esp-idf/components/expat/library/
mkdir -p build//home/lisa/Programming/esp32/esp-idf/components/expat/port/
mkdir -p build//home/lisa/Programming/esp32/esp-idf/components/freertos/
mkdir -p build//home/lisa/Programming/esp32/esp-idf/components/heap/
mkdir -p build//home/lisa/Programming/esp32/esp-idf/components/json/library/
mkdir -p build//home/lisa/Programming/esp32/esp-idf/components/json/port/
mkdir -p build//home/lisa/Programming/esp32/esp-idf/components/log/
mkdir -p build//home/lisa/Programming/esp32/esp-idf/components/lwip/api/
mkdir -p build//home/lisa/Programming/esp32/esp-idf/components/lwip/apps/
mkdir -p build//home/lisa/Programming/esp32/esp-idf/components/lwip/apps/sntp/
mkdir -p build//home/lisa/Programming/esp32/esp-idf/components/lwip/core/
mkdir -p build//home/lisa/Programming/esp32/esp-idf/components/lwip/core/ipv4/
mkdir -p build//home/lisa/Programming/esp32/esp-idf/components/lwip/core/ipv6/
mkdir -p build//home/lisa/Programming/esp32/esp-idf/components/lwip/netif/
mkdir -p build//home/lisa/Programming/esp32/esp-idf/components/lwip/port/freertos/
mkdir -p build//home/lisa/Programming/esp32/esp-idf/components/lwip/port/netif/
mkdir -p build//home/lisa/Programming/esp32/esp-idf/components/mbedtls/library/
mkdir -p build//home/lisa/Programming/esp32/esp-idf/components/mbedtls/port/
mkdir -p build//home/lisa/Programming/esp32/esp-idf/components/newlib/
mkdir -p build//home/lisa/Programming/esp32/esp-idf/components/nghttp/nghttp2/lib/
mkdir -p build//home/lisa/Programming/esp32/esp-idf/components/nghttp/port/
mkdir -p build//home/lisa/Programming/esp32/esp-idf/components/nvs_flash/src/
mkdir -p build//home/lisa/Programming/esp32/esp-idf/components/pthread/
mkdir -p build//home/lisa/Programming/esp32/esp-idf/components/soc/esp32/
mkdir -p build//home/lisa/Programming/esp32/esp-idf/components/spi_flash/
mkdir -p build//home/lisa/Programming/esp32/esp-idf/components/tcpip_adapter/
mkdir -p build//home/lisa/Programming/esp32/esp-idf/components/vfs/
mkdir -p build//home/lisa/Programming/esp32/esp-idf/components/wpa_supplicant/port/
mkdir -p build//home/lisa/Programming/esp32/esp-idf/components/wpa_supplicant/src/crypto/
mkdir -p build//home/lisa/Programming/esp32/esp-idf/components/wpa_supplicant/src/fast_crypto/
mkdir -p build//home/lisa/Programming/esp32/esp-idf/components/xtensa-debug-module/
[... 558 lines cut ...]
CC /home/lisa/Programming/esp32/esp-idf/components/wpa_supplicant/src/crypto/sha1-internal.c
CC /home/lisa/Programming/esp32/esp-idf/components/wpa_supplicant/src/crypto/dh_group5.c
CC /home/lisa/Programming/esp32/esp-idf/components/wpa_supplicant/src/crypto/sha256.c
CC /home/lisa/Programming/esp32/esp-idf/components/wpa_supplicant/src/crypto/rc4.c
CC /home/lisa/Programming/esp32/esp-idf/components/wpa_supplicant/src/crypto/md5.c
[... cut ...]
CC /home/lisa/Programming/esp32/esp-idf/components/wpa_supplicant/src/fast_crypto/fast_sha256.c
CC /home/lisa/Programming/esp32/esp-idf/components/wpa_supplicant/src/fast_crypto/fast_sha256-internal.c
CC /home/lisa/Programming/esp32/esp-idf/components/wpa_supplicant/port/os_xtensa.c
LINK build/application.elf
   text	   data	    bss	    dec	    hex	filename
 687595	 341904	 138176	1167675	 11d13b	build/application.elf
Create build/application.bin
esptool.py v2.1
Create build/firmware.bin
bootloader     15056
partitions      3072
application  1029616
total        1095152

Lisa999, 2017-12-07:
@MrSurly: Basically your saying, start over from scratch... ;-(

MrSurly, 2017-12-07:
@Lisa999 I'm saying make sure the baseline works for you, because we don't have whatever branch you're working from.

If the baseline isn't working, then we can go from there.

Lisa999, 2017-12-07:
Ok, i'll change the baseline. I wish the esp32 was back in the master branch, since i have that also.

nickzoic, 2017-12-07:
Soon :-)

 it does seem rather an odd bug ... there aren't really file permissions
 as such but I think it can throw an EPERM if the parent directory is
 missing or similar ...

dpgeorge, 2017-12-08:
This issue looks somewhat similar to #109.

Lisa999, 2017-12-08:
@dpgeorge as you can see in the used esptool command listed above, i'm using the 0x1000 offset. So i looks like the same problem, but using the offset didn't solve the problem.

MrSurly, 2017-12-08:
@Lisa999 Do you have another device to test on?

Lisa999, 2017-12-08:
** Flashed using different ESP32 (with a oled screen on top of it): **
esptool.py v2.1
Connecting....
Detecting chip type... ESP32
Chip is ESP32D0WDQ6 (revision 1)
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 1091056 bytes to 645654...
Wrote 1091056 bytes (645654 compressed) at 0x00001000 in 16.9 seconds (effective 515.8 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting...

**>>> f = open("main.py", "w")**
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 1] EPERM

So the ESP32 isn't the problem. I'm now replace the upython ESP32 repository.

Lisa999, 2017-12-08:
**esp-idf$ git checkout 2c95a77cf93781f296883d5dbafcdc18e4389656**
fatal: reference is not a tree: 2c95a77cf93781f296883d5dbafcdc18e4389656

?

bboser, 2017-12-08:
FYI: I just ran the same command on a fresh download of the esp-idf. Seems
to work for me (osx).

git checkout 2c95a77cf93781f296883d5dbafcdc18e4389656
>
> M components/bt/lib
>
> M components/esp32/lib
>
> Note: checking out '2c95a77cf93781f296883d5dbafcdc18e4389656'.
>
>
> You are in 'detached HEAD' state. You can look around, make experimental
>
> changes and commit them, and you can discard any commits you make in this
>
> state without impacting any branches by performing another checkout.
>
>
> If you want to create a new branch to retain commits you create, you may
>
> do so (now or later) by using -b with the checkout command again. Example:
>
>
>   git checkout -b <new-branch-name>
>
>
> HEAD is now at 2c95a77c... Merge branch 'bugfix/ioctl_declaration' into
> 'master'
>

MrSurly, 2017-12-08:
> fatal: reference is not a tree: 2c95a77cf93781f296883d5dbafcdc18e4389656
try `git fetch` first

Lisa999, 2017-12-08:
**make erase
make clean
make deploy**
LINK build/application.elf
   text	   data	    bss	    dec	    hex	filename
 699883	 345828	 173520	1219231	 129a9f	build/application.elf
Create build/application.bin
esptool.py v2.1
Create build/firmware.bin
bootloader     15456
partitions      3072
application  1045840
total        1111376
Writing build/firmware.bin to the board
esptool.py v2.1
Connecting.....
Chip is ESP32D0WDQ6 (revision 1)
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 1107280 bytes to 655359...
Wrote 1107280 bytes (655359 compressed) at 0x00001000 in 17.1 seconds (effective 516.6 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting...

**>>> f = open("main.py", "w")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 1] EPERM

**;-(
So i:
Changed ESP32
Updated esp-idf to the right hash
Got me the micropython-esp32 repository (renamed other esp32.old)
Has run make erase, make clean and make deploy
Run the open file command
And still the same error, this error came forward _after_ the bin file crossed the 1024kb file size which might be coincidence or not...

MrSurly, 2017-12-08:
Hmm, the partition size is normally 1MiB, and that might be an issue.  I thought the IDF complained in this case.
Anyway, the compiled binary size I'm getting is 936784

Is your git head @ 84035f0f78b7584961fb7092612194f093188071 ?

Also a long shot: Which xtensa toolchain are you using?
```bash
$ xtensa-esp32-elf-gcc --version
xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-73-ge28a011) 5.2.0
```

Lisa999, 2017-12-08:
I have a LOT of frozen modules, as you can in the code listing in one the posting above.

**xtensa-esp32-elf-gcc --version**
xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-73-ge28a011) 5.2.0

**git rev-parse --short HEAD**
84035f0

MrSurly, 2017-12-08:
> I have a LOT of frozen modules, as you can in the code listing in one the posting above.

Try removing those?

Lisa999, 2017-12-09:
I have finally solved the mystery. I rebuild the esp32/modules redirectory by using a set of modules i use for all platforms. I did not copy _boot.py, initsetup.py and flashbdev.py to the new modules, causing this error.
When i copied these three files over to the new modules directory, everything started to work.
All, thank you for your great support!

MrSurly, 2017-12-09:
> I did not copy _boot.py, initsetup.py and flashbdev.py to the new modules directory, causing this error.

Lisa999, 2017-12-09:
@MrSurly I remember why i had your dev_deepsleep installed: i needed the RTC class... ;-)

MrSurly, 2017-12-10:
@Lisa999 I think you can add it back in, now that the reason for failure is known.

MrSurly, 2017-12-10:
@Lisa999 If this issue has been resolved, please close it.

closed 2017-12-11 by Lisa999

dpgeorge, 2017-12-13:
> I did not copy _boot.py, initsetup.py and flashbdev.py to the new modules directory, causing this error.
> 
> @dpgeorge @nickzoic Can you enlighten us as to why?

Ok, that explains it!  If `_boot.py` does not exist or run then there is no flash filesystem mounted.  Something to remember.