micropython/micropython-esp32 #235: esp_image: Image length 1062848 doesn't fit in partition length 1048576

Lisa999, 2017-12-09:
**E (400) esp_image: Image length 1062848 doesn't fit in partition length 1048576
E (400) boot: Factory app partition is not bootable
E (403) boot: No bootable app partitions in the partition table**

Standard ESP32 code + mrsurly's dev_sleep patch to implement machine.RTC.
This error did NOT happen on the standard ESP32 branch (but then i have  no RTC).

MrSurly, 2017-12-10:
@Lisa999 I _think_ (but I'm not sure) that the newer ESP-IDF actually checks partition length, whereas older ones did not.   IIRC, the partition table wasn't updated for MP, we just moved the flash start sector further out.

In the interim, you can edit `<ESPIDF>/components/partition_table/partitions_singleapp.csv`, and increase the  "factory, app" size from `1m` to `0x180000`.  

@dpgeorge I'll get a PR soon to rectify this.

MrSurly, 2017-12-10:
@Lisa999 Can you test this PR, please?

https://github.com/micropython/micropython-esp32/pull/236

robert-hh, 2017-12-10:
Typo in partitions.csv!

MrSurly, 2017-12-10:
> Typo in partitions.csv!

Yes, there was.  Thanks for the catch @robert-hh.  Fixed now.

Lisa999, 2017-12-10:
@MrSurly: It's working! Thanks a lot! uPyEasy is putting a LOT of demands on micropython, i'm aware of it. The upside is that it's running cross-platform, which is a major benefit.

closed 2017-12-11 by Lisa999

killer-p, 2020-08-10:
> @Lisa999 I _think_ (but I'm not sure) that the newer ESP-IDF actually checks partition length, whereas older ones did not. IIRC, the partition table wasn't updated for MP, we just moved the flash [...]
Can I increase the "factory, app" size from `1M` to a number that larger than 0x18000?Because I still fail even if I increase it to `0x180000`

amolrwagh, 2021-05-12:
Thank you very much ...It is working for me

amolrwagh, 2021-05-12:
> @Lisa999 I _think_ (but I'm not sure) that the newer ESP-IDF actually checks partition length, whereas older ones did not. IIRC, the partition table wasn't updated for MP, we just moved the flash [...]

Working for me

ParameswaranAP, 2022-05-25:
> 

This works for me issue rectified