micropython/micropython-esp32 #239: Can not get ethernet working in olimex ESP32-EVB rev C board

patxitron, 2017-12-11:
I've read the closed issue https://github.com/micropython/micropython-esp32/issues/172 where smeedy got the ethernet working in a ESP32-GATEWAY board from same vendor.

The PHY and GPIOs used are the same in both boards but in mine, following the same steps I'm unable to get the ethernet working (MicroPython v1.9.2-445-g84035f0f on 2017-12-06; ESP32 module with ESP32)

The sequence used is:

```python
>>> import machine
>>> import network
>>> l = network.LAN(mdc = machine.Pin(23), mdio = machine.Pin(18), power = machine.Pin(17), phy_type = network.PHY_LAN8720, phy_addr=0)
I (205443) emac: mac version 1137a
>>> l.ifconfig()
('0.0.0.0', '0.0.0.0', '0.0.0.0', '0.0.0.0')
>>> l.active(1)
I (960717) emac: emac start !!!

I (960717) emac: emac resetting ....
I (960717) emac: emac reset done
I (960717) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
E (961727) emac: Timed out waiting for PHY register 0x2 to have value 0x0007 (mask 0xffff). Current value 0xffff
E (962727) emac: Timed out waiting for PHY register 0x3 to have value 0xc0f0 (mask 0xfff0). Current value 0xffff
E (963727) emac: Timed out waiting for PHY register 0x2 to have value 0x0007 (mask 0xffff). Current value 0xffff
E (964727) emac: Timed out waiting for PHY register 0x3 to have value 0xc0f0 (mask 0xfff0). Current value 0xffff
E (965727) emac: Timed out waiting for PHY register 0x2 to have value 0x0007 (mask 0xffff). Current value 0xffff
E (966727) emac: Timed out waiting for PHY register 0x3 to have value 0xc0f0 (mask 0xfff0). Current value 0xffff
E (967727) emac: Timed out waiting for PHY register 0x2 to have value 0x0007 (mask 0xffff). Current value 0xffff
E (968727) emac: Timed out waiting for PHY register 0x3 to have value 0xc0f0 (mask 0xfff0). Current value 0xffff
E (969727) emac: Timed out waiting for PHY register 0x2 to have value 0x0007 (mask 0xffff). Current value 0xffff
E (970727) emac: Timed out waiting for PHY register 0x3 to have value 0xc0f0 (mask 0xfff0). Current value 0xffff
E (971727) emac: Timed out waiting for PHY register 0x2 to have value 0x0007 (mask 0xffff). Current value 0xffff
```

The last two lines keep repeating forever.

FrankMeyer, 2018-02-21:
Hi Patxitron,
I just know the Problem from the c sdk esp-idf. There was a commit https://github.com/espressif/esp-idf/commit/22ae5adb955d501f4505f4f7ebdf7ad363b5606c that should have fixed the issue, maybe the same can be applyed to the micropython repo.
But for me the Problem still exist. Replugging the USB cable resolves the issue.

dpgeorge, 2018-02-21:
> There was a commit espressif/esp-idf@22ae5ad that should have fixed the issue, maybe the same can be applyed to the micropython repo.

MicroPython was today updated to the latest ESP IDF, see https://github.com/micropython/micropython/commit/c49a73ab0e18035edaa732d31c882a13534caa7a.  This includes the fix mentioned above, so it might work now.

Davidus55, 2018-11-13:
I've also read the closed issue #172.
I don't understand how can I do a "make monitor".
I am on Windows environnement and when I test this code:

```
import machine
import network
lan = network.LAN(mdc = machine.Pin(16), mdio = machine.Pin(17), power = None, phy_type = network.PHY_LAN8720, phy_addr=0)
print('ifconfig :', lan.ifconfig())
print('status : ', lan.status())
print('isconnected : ', lan.isconnected())
print('active : ', lan.active())
```

I have this result:

```
Reading file status
[1/1] Writing file test_lan.py (3kb)
Upload done, resetting board...
OKets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:4732
load:0x40078000,len:7496
load:0x40080400,len:5512
entry 0x4008114c
I (399) cpu_start: Pro cpu up.
I (399) cpu_start: Single core mode
I (399) heap_init: Initializing. RAM available for dynamic allocation:
I (403) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (409) heap_init: At 3FFC4F68 len 0001B098 (108 KiB): DRAM
I (415) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (422) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (428) heap_init: At 40091604 len 0000E9FC (58 KiB): IRAM
I (434) cpu_start: Pro cpu start user code
I (5) cpu_start: Starting scheduler on PRO CPU.
MicroPython v1.9.4-683-gd94aa577a on 2018-11-13; ESP32 module with ESP32
Type "help()" for more information.
>>> Running V:\Informatique\Pyro_ESP32\test_lan.py

>>>
>>> ifconfig : ('0.0.0.0', '0.0.0.0', '0.0.0.0', '0.0.0.0')
status :  None
isconnected :  False
active :  False
>
```

For information I use a ESP32-Gateway Rec C board.

Thank you for your help.

Davidus55, 2018-11-22:
Yes in fact in my example I put another tested line.
But with this command it is the same result.

hagenbuch, 2023-06-15:
I succeeded with 

    lan = network.LAN(mdc = machine.Pin(23), mdio = machine.Pin(18),
                      power = machine.Pin(12), phy_type = network.PHY_LAN8720,
                      phy_addr = 0)

on an Olimex ESP32-POE Rev. B