micropython/micropython-esp32 #243: esp-wrover i2c hardware support

jonhp, 2018-01-08:
hi, i am new here, so if this misses the mark in some way, it is not my intent.
it appears that the esp32-wrover module has full hardware support for i2c.
best i can tell, the micropython implementation of i2c is all in code with 
delays for the clocking.
is there a plan to provide support for the native i2c hardware in micropython?

Avi-TelnT, 2018-01-08:
I believe that ...ports/stm32/machine_i2c.c is HW implemented. and I2C.C with DMA code. 
I see that there is some old CC3000 software code too.  
Search for I2C to see all

MrSurly, 2018-01-08:
This repo has been merged into `micropython/micropython`; issues and PRs for ESP32 (or any other platform) should be posted there.

@jonhp You are correct, software I2C right now.

@Avi-TelnT That's for the STM32 port

Avi-TelnT, 2018-01-08:
@MrSurly
And what about ESP32? 
HW or SW?

MrSurly, 2018-01-08:
@Avi-TelnT Software, right now.  A PR would be welcome, I'm sure.

closed 2019-01-16 by dpgeorge