micropython/micropython-esp32 #244: Is there anyway to debug the booting process?

eminarcissus, 2018-02-07:
I'm now trying to build the image with some basic BLE supports(pull request #86), it does finished compiling but after flashing it to the chip, it won't boots up.

here's the booting message I catched when it is booting.

> picocom v1.7
> 
> port is        : /dev/ttyUSB1
> flowcontrol    : none
> baudrate is    : 115200
> parity is      : none
> databits are   : 8
> escape is      : C-a
> local echo is  : no
> noinit is      : no
> noreset is     : no
> nolock is      : no
> send_cmd is    : sz -vv
> receive_cmd is : rz -vv
> imap is        :
> omap is        :
> emap is        : crcrlf,delbs,
> 
> Terminal ready
> ets Jun  8 2016 00:22:57
> 
> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
> configsip: 0, SPIWP:0xee
> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
> mode:DIO, clock div:2
> load:0x3fff0018,len:4
> load:0x3fff001c,len:7704
> load:0x40078000,len:0
> ho 12 tail 0 room 4
> load:0x40078000,len:13020
> entry 0x4007aeb0
> f)
>   D (42) bm3) boot: SPI Flash Size : 4MB
> D (67) bootloader_flash: mmu set paddr=00000000 count=1
> D (73) boot: mapped partition table 0x8000 at 0x3f408000
> D (78) flash_parts: partition table verified, 3 entries
> I (83) boot: Partition Table:
> I (87) boot: ## Label            Usage          Type ST Offset   Length
> D (94) boot: load partition table entry 0x3f408000
> D (99) boot: type=1 subtype=2
> I (102) boot:  0 nvs              WiFi data        01 02 00009000 00006000
> D (110) boot: load partition table entry 0x3f408020
> D (115) boot: type=1 subtype=1
> I (118) boot:  1 phy_init         RF data          01 01 0000f000 00001000
> D (125) boot: load partition table entry 0x3f408040
> D (13) boot: tge: rea image header @ 0x10000
> D (156) bootloader_flash: mmu set block paddr=0x00010000 (was 0xffffffff)
> D (163) esp_image: image header: 0xe9 0x08 0x02 0x02 40082570
> V (169) esp_image: loading segment header 0 at offset 0x10018
> V (175) esp_image: segment data length 0x53794 data starts 0x10020
> V (181) esp_image: segment 0 map_segment 1 segment_data_offs 0x10020 load_addr 0x3f400020
> I (189) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x53794 (341908) map
> D (198) bootloader_flash: mmu set paddr=00010000 count=6
> V (319) esp_image: loading segment header 1 at offset 0x637b4
> D (319) bootloader_flash: mmu set block paddr=0x00060000 (was 0xffffffff)
> V (320) esp_image: segment data length 0x2a78 data starts 0x637bc
> V (326) esp_image: segment 1 map_segment 0 segment_data_offs 0x637bc load_addr 0x3ffc0000
> I (335) esp_image: segment 1: paddr=0x000637bc vaddr=0x3ffc0000 size=0x02a78 ( 10872) load
> D (344) bootloader_flash: mmu set paddr=00060000 count=1
> V (353) esp_image: loading segment header 2 at offset 0x66234
> D (355) bootloader_flash: mmu set block paddr=0x00060000 (was 0xffffffff)
> V dd 0x40080009otloader_flash: mmu set block paddr=0x00060000 (was 0xffffffff)
> V (403) esp_image: segment data length 0x99cc data starts 0x66644
> V (409) esp_image: segment 3 map_segment 0 segment_data_offs 0x66644 load_addr 0x40080400
> I (417) esp_image: segment 3: paddr=0x00066644 vaddr=0x40080400 size=0x099cc ( 39372) load
> D (426) bootloader_flash: mmu set paddr=00060000 count=2
> V (447) esp_image: loading segment header 4 at offset 0x70010
> D (447) bootloader_flash: mmu set block paddr=0x00070000 (was 0xffffffff)
> V (449) esp_image: segment data length 0xeb358 data starts 0x70018
> V (455) esp_image: segment 4 map_segmen 1 segmets Jun  8 2016 00:22:57
> 
> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
> flash read err, 1000
> ets_main.c 371
> ets Jun  8 2016 00:22:57
> 
> rst:0x10 (RTCWDT_RTC_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))
> waiting for download
> � U
> FATAL: term closed

from flash read err, 1000 I guess it is a bootloader error? But how could I trace down why it did crashed here?

MrSurly, 2018-02-07:
Hi @eminarcissus 

This repo is legacy; the ESP32 port of MP is now part of mainline MP.

BLE support is currently broken.  Also the IDF has progressed quite a bit, and there's a new BLE branch to support that, but I haven't had a lot of extra time to work on that: https://github.com/MrSurly/micropython/tree/dev-bluetooth-esp32

eminarcissus, 2018-02-07:
Hi @MrSurly 

I've noticed the close reason because the IDF change so I merged back bt libs back in 2017 Aug since the remove of osi_arch file. So I guess the bt API part is remained as unchanged. Just curious if there are any way to debug the booting process to see if any error I can resolve so I can get a quick demo on BLE part.
Will look into the new branch as well, thanks very much :)

closed 2018-02-08 by eminarcissus