micropython/micropython-esp32 #246: esp32 uasyncio issue

Sunrise17, 2018-02-11:
Dear @dpgeorge , i have successfully installed 'micropython-uasyncio' lib to my ESP32 board with pip method but now i am receiving following error below as soon as i have called `loop = asyncio.get_event_loop()` method, Why can it caused by?
i am using latest build.
(sysname='esp32', nodename='esp32', release='1.9.3', version='v1.9.3-286-gbbb08431 on 2018-02-09', machine='ESP32 module with ESP32')

    Traceback (most recent call last):
      File "main.py", line 151, in <module>
      File "/lib/uasyncio/core.py", line 224, in get_event_loop
      File "/lib/uasyncio/__init__.py", line 21, in __init__
      File "/lib/uasyncio/core.py", line 30, in __init__
    AttributeError: 'module' object has no attribute 'deque'
    MicroPython v1.9.3-286-gbbb08431 on 2018-02-09; ESP32 module with ESP32

MrSurly, 2018-02-12:
Looks like `asyncio` isn't actually implemented for MP, and only exists to prevent import errors, per the [`setup.py`](https://github.com/micropython/micropython-lib/blob/master/asyncio/setup.py)

robert-hh, 2018-02-12:
@Sunrise17 You have to install uasyncio.queues too. That pulls in collections.

Sunrise17, 2018-02-12:
Dear @robert-hh , I have already installed with pip method `upip.install('micropython-uasyncio.queues')` still i am getting errors in same line. You can see the folder path in ESP32 board in below;

    Kor-MBP:7219_MATRIX kor$ ampy ls /lib/uasyncio
    __init__.py
    core.py
    queues.py

robert-hh, 2018-02-12:
Do you also see lib/collections?

robert-hh, 2018-02-12:
It seems as if the uasyncio files you download with upip are incompatible now with the micropython branch. Older copies of uasyncio file I have here work. So what you can do, is:
clone https://github.com/micropython/micropython-lib
from microypython-lib, copy the files from
uasyncio/uasyncio
uasyncio.core/uasyncio
uasyncio.queues/uasyncio
uasyncio.synchro/uasyncio
to a directory called uasyncio on your device (could a well be lib/uasyncio). After that, the local uasyncio directory should contains the files:
__init__.py
core.py
queues.py
synchro.py

Sunrise17, 2018-02-12:
Dear @robert-hh , thanks for your solution, and it works well now without an issue.

closed 2018-02-12 by Sunrise17