micropython/micropython-esp32 #248: unable to connet to esp32 from webREPL

sanu-krishnan, 2018-07-12:
This happens when there is no delay statement in the infinite while loop or the delay is too small

```
while True:
      led.value(1)
      time.sleep_ms(10)
      led.value(0)
      time.sleep_ms(10)
```

Above code has the issue. But when I change the delay values to 100, it works fine.

**Full Code : ebotmain.py**
```
import time
import machine
def ebotx():	
	led=machine.Pin(5,machine.Pin.OUT)
	while True:
		led.value(1)
		time.sleep_ms(10)
		led.value(0)
		time.sleep_ms(10)
```

**boot.py**
```
import network
import sys
import webrepl
import time

sta_if = network.WLAN(network.STA_IF)
sta_if.active(True)
sta_if.ifconfig(('192.168.1.200','255.255.255.0','192.168.1.1', '8.8.8.8'))
sta_if.connect('xxxxx', 'xxxxxx')
sta_if.isconnected()

webrepl.start()
import ebotmain
del sys.modules['ebotmain']
import ebotmain
ebotmain.ebotx()
```

Avi-TelnT, 2018-07-12:
In the 2 codes the delay is the same. 10 Msec…. 

To make infinity loop is bad programing. Use timer interrupt for that.

 

-Avi

sanu-krishnan, 2018-07-12:
yea. both are the same code. 2nd and 3rd are the complete code.

The code inside the while loop won't be same as this blink code, and I need to repeat it till 'ebotmain.py' is updated.

According to my knowledge, webREPL can run as a daemon (in the background) on esp32 and it uses a dedicated hook to run periodically in the main uPy task in order to check for incoming connections.

Should I set any parameters to use webREPL in daemon mode?

Avi-TelnT, 2018-07-12:
Why not to use timer event to blink the led? 

And kill the timer when you do not need it. 

It has nothing with webREPL or other task you need blinking led. As Wi-Fi connection etc. 

-Avi

MrSurly, 2018-07-12:
> Why not to use timer event to blink the led?

This doesn't fix the core issue of web REPL being unresponsive when there's a tight loop.

MrSurly, 2018-07-12:
@sanu-krishnan This repo is considered deprecated as the ESP32 port is now part of the [main repo](https://github.com/micropython/micropython).  You should open an issue there, with the title specifying that this is an ESP32 e.g. `ESP32: unable to connect from webREPL`

Avi-TelnT, 2018-07-13:
No need any tight loop anywhere. Make the ESP32 stress is not good for other tasks too. 

You can call my Skype nissim.test

I am Avi boss. 

-Nissim

sanu-krishnan, 2018-07-13:
@MrSurely : sure.

I got a reply from Mr.dpjeorge saying that 20ms delay is compolasry for the present implementation

sanu-krishnan, 2018-07-23:
Thanks nickzoic,
LED pulsing is just an example. My idea is to use WebREPL while execution of the present code, to pause that code and update the same code.
I managed it somehow by adding an interrupt on pin0, (ISR includes some delay) and pressing the button connected to pin0 to access the webREPL.

Will report further issues in the main repo.

Thank you all.

closed 2018-07-23 by sanu-krishnan