micropython/micropython-esp32 #252: Is it really RANDOM ?

curlyz, 2018-10-22:
Hi 
I am using 
**(sysname='esp32', nodename='esp32', release='1.9.4', version='v1.9.4-644-g7de921
1b8 on 2018-10-13', machine='ESP32 module with ESP32')**

Just doing some neopixel ring , 12 RGB . 

```python
from machine import *
from neopixel import *
from time import *
from random import randrange

n = NeoPixel(Pin(5) , 12 , timing = True)

for x in range(12):
  n[x] = (randrange(50),randrange(50),randrange(50))
  n.write()
  sleep_ms(10)
```

So . The weird thing is that whenever I press reset or plug in the USB , it show the same color , shouldn't it be random ?

Avi-TelnT, 2018-10-22:
If you drive the neopixel with VCC=5V and ESP32 IO with 3V3 expect errors. 

You can call my Skype nissim.test

dpgeorge, 2018-10-22:
>  shouldn't it be random ?

The random module is pseudo random and requires a seed value.  Upon hard-reset the seed will be the same each time, so the sequence will be the same.  To get different values you can seed it with the clock (assuming that changes), or use `os.urandom()` for true random numbers.

closed 2018-10-22 by dpgeorge