micropython/micropython-esp32 #257: AT command connected user number ESP32 devkitc V4

hiflord, 2020-01-31:
Hello. I am doing a project with esp32 devkitc v4. I send AT commands to setup webserver and when one user connects my web server, his user number is coming to me like 1,CONNECT. After that I am using that user number for closing server. The sample picture is in below.
[image]
Sometimes two connect number is coming sometimes one connect number is coming. I didnt understand how is work the system. The sample picture two is in below.
[image]
and thinking algorithmically seems difficult to me. That is the question Is there any AT command to learn which user connected my web server? Thank you help me btw.

nickzoic, 2020-01-31:
Hi!  This is the old repo for MicroPython on ESP32 ... a Python which runs right on your ESP32.
It's closed now as this project has been merged into the main https://github.com/micropython/micropython/ repository.

I think you're asking about the AT commands firmware for ESP32, which is a whole different thing which can be found over here: https://github.com/espressif/esp-at ... I don't know anything much about it but probably the best place to start is https://github.com/espressif/esp-at/blob/master/docs/ESP_AT_Commands_Set.md

closed 2020-01-31 by nickzoic

hiflord, 2020-01-31:
Thankyou. Sory for my mistake.