micropython/micropython-esp32 #261: recipe for target 'build-GENERIC_SPIRAM/bootloader.elf' failed

MRxing1, 2021-04-13:
I'm compiling the firmware of esp32. I encountered the following problems and asked for assistance；

xkp@xkp-VirtualBox:~/esp32/micropython/ports/esp32$ make
MicroPython v1.13-30-g1d344f446-dirty
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
Building with ESP IDF v4
LINK build-GENERIC_SPIRAM/bootloader.elf
/home/xkp/esp32/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find -lrtc
collect2: error: ld returned 1 exit status
Makefile:979: recipe for target 'build-GENERIC_SPIRAM/bootloader.elf' failed
make: *** [build-GENERIC_SPIRAM/bootloader.elf] Error 1

xkp@xkp-VirtualBox:~/esp32/micropython/ports/esp32$ make V=1
MicroPython v1.13-30-g1d344f446-dirty
Building with ESP IDF v4
LINK build-GENERIC_SPIRAM/bootloader.elf
xtensa-esp32-elf-gcc  -nostdlib -L/home/xkp/esp32/esp-idf/lib -L/home/xkp/esp32/esp-idf/ld -u call_user_start_cpu0 -Wl,--gc-sections -static -Wl,-EL -Wl,-Map=build-GENERIC_SPIRAM/bootloader.map -Wl,--cref -T /home/xkp/esp32/esp-idf/components/bootloader/subproject/main/esp32.bootloader.ld -T /home/xkp/esp32/esp-idf/components/bootloader/subproject/main/esp32.bootloader.rom.ld -T /home/xkp/esp32/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld -T /home/xkp/esp32/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld -T /home/xkp/esp32/esp-idf/components/esp32/ld/esp32.peripherals.ld -o build-GENERIC_SPIRAM/bootloader.elf  -Wl,--start-group  -Lbuild-GENERIC_SPIRAM/bootloader -lbootloader_support -llog -lspi_flash -lsoc -lmain -L/home/xkp/esp32/esp-idf/components/esp_wifi/lib_esp32 -lrtc -L/home/xkp/esp32/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/esp32-psram/ -lgcc -Wl,--end-group
/home/xkp/esp32/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find -lrtc
collect2: error: ld returned 1 exit status
Makefile:979: recipe for target 'build-GENERIC_SPIRAM/bootloader.elf' failed
make: *** [build-GENERIC_SPIRAM/bootloader.elf] Error 1

Hope to get your help, thank you very much

MRxing1, 2021-04-13:
From the beginning of the compilation log, the feeling is a little bit worse, because makefile has reached the end：

xkp@xkp-VirtualBox:~/esp32/micropython/ports/esp32$ make
MicroPython v1.13-30-g1d344f446-dirty
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
Building with ESP IDF v4
GEN build-GENERIC_SPIRAM/sdkconfig.h
mkdir -p build-GENERIC_SPIRAM/bootloader//home/xkp/esp32/esp-idf/components/bootloader/subproject/main/
mkdir -p build-GENERIC_SPIRAM/bootloader//home/xkp/esp32/esp-idf/components/bootloader_support/src/
mkdir -p build-GENERIC_SPIRAM/bootloader//home/xkp/esp32/esp-idf/components/bootloader_support/src/esp32/
mkdir -p build-GENERIC_SPIRAM/bootloader//home/xkp/esp32/esp-idf/components/log/
mkdir -p build-GENERIC_SPIRAM/bootloader//home/xkp/esp32/esp-idf/components/soc/esp32/
mkdir -p build-GENERIC_SPIRAM/bootloader//home/xkp/esp32/esp-idf/components/soc/src/
mkdir -p build-GENERIC_SPIRAM/bootloader//home/xkp/esp32/esp-idf/components/spi_flash/
CC /home/xkp/esp32/esp-idf/components/bootloader_support/src/bootloader_clock.c
CC /home/xkp/esp32/esp-idf/components/bootloader_support/src/bootloader_common.c
CC /home/xkp/esp32/esp-idf/components/bootloader_support/src/bootloader_flash.c
CC /home/xkp/esp32/esp-idf/components/bootloader_support/src/bootloader_flash_config.c
CC /home/xkp/esp32/esp-idf/components/bootloader_support/src/bootloader_init.c
CC /home/xkp/esp32/esp-idf/components/bootloader_support/src/bootloader_random.c
CC /home/xkp/esp32/esp-idf/components/bootloader_support/src/bootloader_utility.c
CC /home/xkp/esp32/esp-idf/components/bootloader_support/src/flash_qio_mode.c
CC /home/xkp/esp32/esp-idf/components/bootloader_support/src/esp_image_format.c
CC /home/xkp/esp32/esp-idf/components/bootloader_support/src/flash_encrypt.c
CC /home/xkp/esp32/esp-idf/components/bootloader_support/src/flash_partitions.c
CC /home/xkp/esp32/esp-idf/components/bootloader_support/src/esp32/bootloader_sha.c
CC /home/xkp/esp32/esp-idf/components/bootloader_support/src/esp32/secure_boot.c
AR build-GENERIC_SPIRAM/bootloader/libbootloader_support.a
CC /home/xkp/esp32/esp-idf/components/log/log.c
AR build-GENERIC_SPIRAM/bootloader/liblog.a
CC /home/xkp/esp32/esp-idf/components/spi_flash/spi_flash_rom_patch.c
AR build-GENERIC_SPIRAM/bootloader/libspi_flash.a
CC /home/xkp/esp32/esp-idf/components/soc/esp32/cpu_util.c
CC /home/xkp/esp32/esp-idf/components/soc/esp32/gpio_periph.c
CC /home/xkp/esp32/esp-idf/components/soc/esp32/rtc_clk.c
CC /home/xkp/esp32/esp-idf/components/soc/esp32/rtc_clk_init.c
CC /home/xkp/esp32/esp-idf/components/soc/esp32/rtc_init.c
CC /home/xkp/esp32/esp-idf/components/soc/esp32/rtc_periph.c
CC /home/xkp/esp32/esp-idf/components/soc/esp32/rtc_pm.c
CC /home/xkp/esp32/esp-idf/components/soc/esp32/rtc_sleep.c
CC /home/xkp/esp32/esp-idf/components/soc/esp32/rtc_time.c
CC /home/xkp/esp32/esp-idf/components/soc/esp32/rtc_wdt.c
CC /home/xkp/esp32/esp-idf/components/soc/esp32/sdio_slave_periph.c
CC /home/xkp/esp32/esp-idf/components/soc/esp32/sdmmc_periph.c
CC /home/xkp/esp32/esp-idf/components/soc/esp32/soc_memory_layout.c
CC /home/xkp/esp32/esp-idf/components/soc/esp32/spi_periph.c
CC /home/xkp/esp32/esp-idf/components/soc/src/memory_layout_utils.c
AR build-GENERIC_SPIRAM/bootloader/libsoc.a
CC /home/xkp/esp32/esp-idf/components/bootloader/subproject/main/bootloader_start.c
AR build-GENERIC_SPIRAM/bootloader/libmain.a
LINK build-GENERIC_SPIRAM/bootloader.elf
/home/xkp/esp32/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find -lrtc
collect2: error: ld returned 1 exit status
Makefile:979: recipe for target 'build-GENERIC_SPIRAM/bootloader.elf' failed
make: *** [build-GENERIC_SPIRAM/bootloader.elf] Error 1

This is the execution order：make V=1 Print information at the back

nents/esp_wifi/include -I/home/xkp/esp32/esp-idf/components/esp_wifi/esp32/include -I/home/xkp/esp32/esp-idf/components/lwip/include/apps/sntp -I/home/xkp/esp32/esp-idf/components/spi_flash/private_include -I/home/xkp/esp32/esp-idf/components/wpa_supplicant/include/esp_supplicant -I/home/xkp/esp32/esp-idf/components/xtensa/include -I/home/xkp/esp32/esp-idf/components/xtensa/esp32/include -I/home/xkp/esp32/esp-idf/components/bt/include -I/home/xkp/esp32/esp-idf/components/bt/common/osi/include -I/home/xkp/esp32/esp-idf/components/bt/common/btc/include -I/home/xkp/esp32/esp-idf/components/bt/common/include -I/home/xkp/esp32/esp-idf/components/bt/host/nimble/nimble/porting/nimble/include -I/home/xkp/esp32/esp-idf/components/bt/host/nimble/port/include -I/home/xkp/esp32/esp-idf/components/bt/host/nimble/nimble/nimble/include -I/home/xkp/esp32/esp-idf/components/bt/host/nimble/nimble/nimble/host/include -I/home/xkp/esp32/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/ans/include -I/home/xkp/esp32/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/bas/include -I/home/xkp/esp32/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/gap/include -I/home/xkp/esp32/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/gatt/include -I/home/xkp/esp32/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/ias/include -I/home/xkp/esp32/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/lls/include -I/home/xkp/esp32/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/tps/include -I/home/xkp/esp32/esp-idf/components/bt/host/nimble/nimble/nimble/host/util/include -I/home/xkp/esp32/esp- [... cut ...] log -lspi_flash -lsoc -lmain -L/home/xkp/esp32/esp-idf/components/esp_wifi/lib_esp32 -lrtc -L/home/xkp/esp32/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/esp32-psram/ -lgcc -Wl,--end-group
/home/xkp/esp32/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find -lrtc
collect2: error: ld returned 1 exit status
Makefile:979: recipe for target 'build-GENERIC_SPIRAM/bootloader.elf' failed
make: *** [build-GENERIC_SPIRAM/bootloader.elf] Error 1

dpgeorge, 2021-04-13:
This repository is now obsolete.  Please use https://github.com/micropython/micropython

closed 2021-04-13 by dpgeorge

MRxing1, 2021-04-13:
> This repository is now obsolete. Please use https://github.com/micropython/micropython

Hello, thank you very much for your reply!

I would like to point out that although this is an old version of microppython, I transferred the version of microppython I used from GitHub to gitee, and made a lot of optimizations. Therefore, if I use the latest version, I still need to re transfer and re add sub modules, so could you please analyze it? Thank you very much!