micropython/micropython-esp32 #266: Make UART allow Pin

MATTYGILO, 2022-11-18:
It would be great if UART accepted Pin class instead of integers
```
>>> UART(1, 9600, timeout=1000, rx=Pin(18), tx=Pin(17))
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: can't convert Pin to int
```

BEMNETADEFIS, 2022-11-19:
The micropython reads and also combines concept from C programming 2 python
allocation of memory and registers in C using hexadecimal combined with
variables R known stratgy for hardware programming so if u can use this
techniques for programming ur module with less lines of conjugation of line
of array so that u can locate UART registers for managing UART port
Also python hv options for this i think

nickzoic, 2022-11-21:
I kind of agree with you there @MATTYGILO but please note this repository is no longer active, the ESP32 port has been merged into the main Micropython repo at https://github.com/micropython/micropython